motioneye-project/motioneyeos #1873: WiFi error with Pi zero

EowynJim, 2019-03-26:
I have a new Pi Zero W and have successfully loaded the full raspian code and connected to my local WiFi.  Have been unsuccessful in efforts to load MotionEyeOS either the 20190119 or 20190321 version.  I'm getting network failure from the Broadcom load.  I've rewritten the wpa_supplicant file several times.  With my SSID="EowynHome_5GHZ" things just fail.  With a "EowynGuest" network the card failed to pull a DHCP address.  Is there a way to add MotionEye as an application to the SD card that I have working with raspian? I'm using Windows 10 on a SurfacePro and I used SD Formatter, Etcher and Notepad++.

ccrisan, 2019-03-26:
@EowynJim I'm sorry your experience with motionEyeOS isn't as smooth as expected. If you want to use motionEye on top of Raspbian, just use [these](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) instructions.

closed 2019-03-26 by ccrisan

user9023523, 2019-04-07:
Pi Zero W doesn't have 5 GHz networking.