motioneye-project/motioneyeos #1875: Dev20190327 fnc not working system monitoring?

tigernero79, 2019-03-27:
@ccrisan Do you also confirm that in fast Bot mode, even if you enable the voice system monitoring, the FPS item does not exit by clicking on image? in motioneye mode it works instead. this release however solved the problem of the update from previous versions. thank you

ccrisan, 2019-04-01:
@tigernero79 I'm not sure I understand what you mean. Please make sure you're using the correct names of the various functions of motionEye(OS).

I assumed that *fast boot camera mode* would in fact mean *fast network camera mode*, but the *the voice system monitoring* beats me.

Please attach a couple of screenshots exemplifying what you mean by *FPS item does not exit by clicking on image*.

tigernero79, 2019-04-01:
@ccrisan 
Example working button enable system monitoring motioneyeos

[image]

Example not working the button enable system monitoring on fast network camera enable
[image]

tigernero79, 2019-04-02:
@ccrisan 

in practice it does not take into account the file monitor_1 and not from values ​​by clicking on video, also with dev20190402 I noticed that ilf ile monitor_1 exists but now neither in motioneyos than in fastboot camera mode from values. does not take into account the file monitor_1 in short

ccrisan, 2019-04-02:
The fast network camera mode is a simple MJPEG stream directly displayed in browser. There's no (easy) way we can transmit monitoring information with MJPEG frames. And this is not some regression bug recently introduced.

closed 2019-04-02 by ccrisan

tigernero79, 2019-04-02:
@ccrisan ccrisian with dev20190402 even in normal mode now I have more info from monitor_1, so I don't talk about fcn now, in practice I no longer see the values ​​in degrees fps from monitor_1, this with your defaul script, you see them?