motioneye-project/motioneyeos #1876: Camera is broken after update

johny-mnemonic, 2019-03-29:
I just completely lost two of my cameras after upgrading them to latest stable FW. One was internal and easily reachable, so I brought it to my man cave just to find out, it is in a boot loop spitting following error:
VFS: Unable to mount root fs

After some googling I have found, that it is known issue with fwupdater, which is offering update from versions older than 20181129, while it is not possible to upgrade from these.

Could you please fix this asap, as lot of people can get caught like me?

It is really a big issue to let it in this state, when automatic updater will completely brick the device. My second broken camera is high on the house wall, so I am not looking forward getting there twice...

It is nice, that there is a bold note in release notes with a link to an experimental way of manual upgrade, but none of these is visible from the webUI of the camera, from where people usually do updates.  So the updater still goes and bricks cameras, while it can't be hard to add check, that will not allow such upgrade if the running version is not compatible with the new FW...
It will save a lot of people big troubles

tigernero79, 2019-04-01:
https://github.com/ccrisan/motioneyeos/issues/1878#issuecomment-478533565

jasaw, 2019-04-01:
@johny-mnemonic I agree with you that a lot of people will be caught out. See our discussion here: https://github.com/ccrisan/motioneyeos/issues/1803

johny-mnemonic, 2019-04-02:
@jasaw I just went through that discussion and was surprised by the result.
Then I checked the code of fwupdate script where I finally understood why it ended that way. There was exactly zero checks, before the flashboot function was called, in old version of fwupdate :-( 
That's really sad. Not even hash verification, no nothing :-(  I did't expect that in a script, that is flashing OS.
Without any kind of check in place, it is hard to come up with anything, that could possibly save users the headache of dead devices after upgrade.

The only thing you can do in such scenario is to change the file extension of new versions. This would be the only way to tell the old fwupdate script to fail when it downloads incompatible FW.

stefi01, 2019-08-03:
i am using version...

motionEye Version | 0.39.3
-- | --
Motion Version | 4.1.1
OS Version | Raspbian 9.4
raspberry pi 3b

 went to upgrade and got a whole bunch of stuff thrown at me

Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/51/fe/18125dc680720e4c3086dd3f5f95d80057c41ab98326877fc7d3ff6d0ee5/Pillow-6.1.0.tar.gz (33.8MB)
    99% |████████████████████████████████| 33.8MB 1.3MB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 882, in _download_http_url
    _download_url(resp, link, content_file, hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 603, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/lib/python2.7/dist-packages/pip/utils/hashes.py", line 46, in check_against_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 571, in written_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/utils/ui.py", line 139, in iter
    for x in it:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 560, in resp_read
    decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/response.py", line 432, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/response.py", line 380, in read
    data = self._fp.read(amt)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 63, in read
    self._close()
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 87, in dumps
    ).encode("utf8"),
MemoryError

stefi01, 2019-08-03:
when i try a more updated version of raspbian then motioneye opens but no camera