motioneye-project/motioneyeos #1877: Check for update reports it’s up to date but it’s not

ThomastheBicyclist, 2019-04-01:
OS version “dev20190126” reports up to date message when software update: “check” button is pressed but I see dev20190327 is available on the repo. 

motionEye version: 0.39.3
Motion version: 4.1.1
OS Version: motionEyeOS dev20190126

hardware: raspberry pi nano w
overclocking: off

tigernero79, 2019-04-01:
The enable pre-release button is active ?, however I suggest you do clean installation of dev20190327 this because update can lead you to no longer display the picamera if you have mounted one.

ccrisan, 2019-04-01:
@ThomastheBicyclist there are basically 3 types of motionEyeOS releases:

1. stable releases (e.g. `20190230`, marked as stable on GitHub)
2. prereleases (e.g. `20190230`, marked as prerelease on GitHub)
3. nightly builds (e.g. `dev20190230`, marked as prerelease on GitHub)

If you're on the nightly "channel", I recommend upgrading manually using `fwupdate upgrade <version>`.
If you prefer the prereleases "channel", then @tigernero79's answer is the correct answer.
If you prefer using stable releases, then do not use a nighty build :)

closed 2019-04-01 by ccrisan