motioneye-project/motioneyeos #1878: Strange camera artifact on Foscam camera

gezuka77, 2019-04-01:
Hi,

First issue post on Github. I try to post just the issue and try to get away without posting logs...
[image]
I am running RPI 3 B+. Other two cameras are working fine. They are not identical ones.
I have tried with/ without the motioneyeos overlay, but problem is the same.
The artifacts are going away sometimes and also not on a constant places. Sometimes 1-2 lines sometimes half the image.

Ideas?

tigernero79, 2019-04-01:
what version do you use motioneye? have you tried to cleanly install the 20190327 version?

gezuka77, 2019-04-01:
Hi, I am on motion
Eye Version	0.40rc5
Motion Version	4.2.2
OS Version	motionEyeOS dev20190321

tigernero79, 2019-04-01:
have you tried to cleanly install the 20190327 version?

ccrisan, 2019-04-01:
@gezuka77 does the problem appear also when you're viewing the MJPEG stream directly? (e.g. http://ip:8081).

gezuka77, 2019-04-02:
Hi, yes it seems to be there in live view/recording and also in streaming from MotioneyeOs...
The artifact comes and goes and also not the same all the time, sometimes takes up almost half the screen.

ccrisan, 2019-04-02:
@gezuka77 then it's probably not a motionEye bug.

gezuka77, 2019-04-02:
It is a clean install so something is clearly not good.. Should I post some error reports?
Also, I have an Ubuntu media server and I put MotionEye on that, that does not produce the same artifacts.

ccrisan, 2019-04-02:
You could try adding your network camera as TCP instead of UDP, if we're talking about a RTSP camera. I don't have any other ideas. If you are sure this is not a problem of your camera (i.e. the stream looks good on other programs such as VLC), then you can [report a Motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).