motioneye-project/motioneyeos #1880: don't advertise updates in WEB GUI if they cannot be applied automatically

ralfixx, 2019-04-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board Model

I am using the following board/model:  Raspberry PI Zero W

### Camera

I am using the following type of camera: Fast Network Camera

My camera model is:  Raspberry CSI camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

NONE

### Log Files

I consider the following log files relevant to this issue:

NONE

--------------------

MotionEyeOS should not advertise Updates in the WEB-GUI which cannot be applied automatically.   

The latest upgrades all read 

"note: do not attempt to update to this version automatically! "
  
However this is not reflected in the WEB interface which simply says "Update available" when checking.  If you then apply the Update, the device is rendered unusable, requiring physical access to the device to recover which in some cases can be problematic (e.g. remote site not easily accessible).

At least there should be a warning in the WEB-Interface about the update not being suitable for automatic updates.

Thank you for MotionEyeOS!

ccrisan, 2019-04-08:
@ralfixx I am sorry about your inconvenience. I really am. We've tried many approaches to make this update smooth but unfortunately we had no possible way of automatically reorganizing the partition table (which is what happened with recent versions).

Starting with these recent versions, we have now the possibility to prevent an update in case of incompatibilities and, basically, we have more control on the update process.

Again, we are sorry for this problem and for not being able to automatize or prevent this particular upgrade.

closed 2019-04-08 by ccrisan

ralfixx, 2019-04-09:
* Calin Crisan
| @ralfixx I am sorry about your inconvenience. I really am. We've
| tried many approaches to make this update smooth but unfortunately
| we had no possible way of automatically reorganizing the partition
| table (which is what happened with recent versions).
| 
| Starting with these recent versions, we have now the possibility to
| prevent an update in case of incompatibilities and, basically, we
| have more control on the update process.
| 
| Again, we are sorry for this problem and for not being able to
| automatize or prevent this particular upgrade.

Crisan,
no problem here, the device was located in a birds' nesting box,
and they hadn't started breeding yet, so I could swap the SD-cards
without disturbance.

Thank you very much again for all you effort on MotionEyeOS!

R'