motioneye-project/motioneyeos #1882: Restore

vplustute, 2019-04-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20190402

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: KKMOON

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet)

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

I try the new version 
motionEye Version | 0.40rc5
Motion Version | 4.2.2
OS Version | motionEyeOS dev20190402

bevor I chanced I made a backup from my configuration. I have 4 network cameras
But when I try to restore all the Settings, it doesn't work. I get message "The configuration has been restored!" but there is nothing restored.

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3035608/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3035609/boot.log)

ccrisan, 2019-04-02:
Have you restored from a previous version?

vplustute, 2019-04-03:
Yes, I made a backup from the Version
motionEye Version
0.39.3
Motion Version
4.1.1
OS Version
motionEyeOS dev20190203

After that I set up the Version
motionEye Version | 0.40rc5
Motion Version | 4.2.2
OS Version | motionEyeOS dev20190402

and tried to restore all the Settings. I thought at this way I don't have to configurate all my cameras new.

ccrisan, 2019-04-03:
You won't be able to do that since many configuration options have changed. Please start configuring your new motionEyeOS from scratch.

closed 2019-04-03 by ccrisan

Instagit, 2019-04-28:
@ccrisan If the new version is not compatible anymore, why not make a release that disables auto update for the old versions first to avoid breaking everyone's install? Pretty disappointing.

ccrisan, 2019-04-28:
@Instagit because people with older versions would eventually upgrade to the real new version any way, skipping any intermediate versions (that's how the upgrade worked in previous versions).

Instagit, 2019-04-29:
@ccrisan
>people with older versions would eventually upgrade to the real new version any way

Absolutely, but being forced to reinstall by a broken upgrade on a sunday evening is not very user friendly. Some people rely on motionEyeOS for security etc.
This way you are conditioning users to never install updates, because they might break their system. :stuck_out_tongue:

xMTinkerer, 2019-05-02:
Ok, so I'm running motioneye in a docker container and I realized I needed to expose the streaming ports. Not being an expert in docker, I took a backup from the motioneye UI, terminated the container, removed the container and image, updated my `docker run` command and fired it. 
Apparently the container I was using was rather old (> 6 months) and I now have a new version of motioneye running. 
When I got into motioneye, I had no cameras. Ok rather expected as I removed everything, but then when I went to restore my backup, there was no error message and instead the call to action to add a new camera. So now I have to go and re-add my cameras with all the settings I had just right to the way I wanted them just because I upgraded. 
So I guess my question is, what is the proper process for upgrading when running docker? But to @Instagit's point not being able to backup and restore across versions is not really conducive to ever upgrading.

ccrisan, 2019-05-02:
@xMTinkerer backing up and restoring between different versions of motionEye usually works. It doesn't when there are significant changes in the config files (such as 0.39 - 0.40). It also doesn't work between motionEye and motionEyeOS.