motioneye-project/motioneyeos #1883: Only one of two cameras works. The other always shows "Unable To Open Video Device"

TheJWays, 2019-04-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: ALLOYSEED 12MP. This is a simple chinese Webcam which can be bought here:

https://www.aliexpress.com/item/Newest-Webcam-USB-12-Megapixel-High-Definition-Camera-Web-Cam-360-Degree-MIC-Clip-on-For/32898640018.html?spm=a2g0s.9042311.0.0.27424c4dx8inDy

I am using two exactly identical models.

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

### Peripherals
I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 [motion.log](https://github.com/ccrisan/motioneyeos/files/3044704/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3044705/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3044706/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3044707/dmesg.log)

Good day. First of all: Thank you for making such a great and easy to configure/use surveilance software. Unfortunately, there seems to be an issue with the loading of the cameras: One of the two cameras is always displayed correctly and the other one has a gray picture with a text saying "Unable to open video device". There are a few things that are intresting about this:

First of all, when loading a page, there is a small change of both cameras being displayed correctly. However, the stream is stopped and the gray error message is displayed shortly after that.

Secondly: The working camera is randomly displayed as either camera1 or camera2 upon loading the page. It seems like the physical camera is not directly linked to the virtual camera1/2 instance.

Last of all: The "Camera Device" Setting shows two very different things for the two camera instances:

Camera1 displays "/dev/v4l/by-id/usb-Generic_USB2.0_PC_CAMERA-video-index0"
Camera2 displays "/dev/video1"

Under /dev, there are the video-devices 

video1
video2

Under /dev/v4l/by-Path there are the entries

platform-3f980000.usb-usb-0:1.1.3:1.0-video-index0@ 
platform-3f980000.usb-usb-0:1.2:1.0-video-index0@

Thank you very much for your feedback/support.

ccrisan, 2019-04-08:
This sounds like a bandwidth problem. Or a USB power problem. Or the driver is so badly written that it can't handle two identical devices. I would start by reducing the framerate/resolution and seeing if the problem persists.

TheJWays, 2019-04-08:
Thanks for your answer. I have tested all the points:
- Power: I measured the 5V-USB-Current consumed by the cameras. It is as little as 8mA while working. This is absolutely no problem for the 3A PSU.
- Bandwidth: The System operates in a gigabit Networking environment. Nevertheless, I have reduced the Framerate and the resolution of both cameras to minimum and the problem still persists.
- Driver: That is possible. I do not know how to check/fix that though. Isn't the default V4L2-Driver used to read the cameras?

I think the behaviour of the correct camera being randomly displayed as either camera1 or camera2 combined with the different "Camera Device" Settingvalues indicates a Software/Driver-Issue with the camera...

TheJWays, 2019-04-08:
Do the Log-Files i've posted contain anything meaningful to this issue?

jasaw, 2019-04-09:
@TheJWays From `motion.log`, I think motion has been configured to read both streams from the same camera.
>v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy

You have 2 identical cameras, so you end up with `/dev/video1` and `/dev/video2`. If you modify the `/data/etc/thread-1.conf` and `/data/etc/thread-2.conf` files to use `/dev/video1` and `/dev/video2` respectively, it should work.

I suspect the reason why camera by-id does not work is because this cheap Chinese camera does not contain a unique USB ID, so the by-id path `/dev/v4l/by-id/usb-Generic_USB2.0_PC_CAMERA-video-index0` gets clobbered when you have more than 1 camera plugged in. This is a violation of the USB standard by the Chinese manufacturer.

TheJWays, 2019-04-09:
@jasaw 
Wow, that was spot on. Using lsusb, I could indeed determine that both of the USB-ID strings are identical. Well thats what I get for buying cheap chinese cameras...
Editing the "videodevice"-Setting in the two thread.conf files to the absolute device path did the trick. I thank you very much for your assistance and wish you a pleasant day.

closed 2019-04-09 by TheJWays