motioneye-project/motioneyeos #1885: notification delay

martin1957, 2019-04-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

martin1957, 2019-04-06:
I am using Pi Zero-W with a small python script just to output a puls on a GPIO.
This pulse is used to trigger a DSLR camera for higher picture quality on motion.
Everything works fine so far.
The only problem is the delay of detected motion to the trigger-pulse output.
I see on the picture-screen that motion is detected. Then I have a delay between 5 to 10 seconds 
until the pulse is sent.
Is there any possibility to reduce this delay ??

user9023523, 2019-04-08:
You need a faster board unfortunately. I notice the same lag on my Zero W but not my 3B+.

martin1957, 2019-04-08:
Yes, I saw the same. Due to power consumption I cannot use the 3B+.
I build a little more complex system to use outdoors (batterie supply) for amimal photographie which is based on a ESP32 controller to move my DSLR via steppers on a tripod. The Zero-W with MotionEyeOS  is more or less just a blackbox for me to create the first triggerpulse for the DSLR on any movement.
I will now test elder versions of MotionEyeOS  to see if there is any difference in the lag.

user9023523, 2019-04-08:
Try taking a look at this project if your motion cam is connected with ribbon cable.
https://elinux.org/RPi-Cam-Web-Interface