motioneye-project/motioneyeos #1887: Unable to unzip the *img.xz file on a Mac

duncangroenewald, 2019-04-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

duncangroenewald, 2019-04-08:
I just downloaded one of the images to a Mac (10.14.4) and don't seem to be able to unzip the *.xz file.  What is the *.xz file format - I assumed it was just a ZIP archive but changing the extension to zip does'nt seem to help.  I can't help thinking I must be doing something stupid here.

jasaw, 2019-04-09:
You can't unzip xz files because xz is not zip. For more info about XZ format, read this: https://en.wikipedia.org/wiki/Xz

On a mac, there are lots of ways to decompress an xz file. I found this from google search: https://www.quora.com/How-do-you-open-extract-a-tar-xz-file-on-Mac-OS-X

closed 2019-04-10 by duncangroenewald