motioneye-project/motioneyeos #1889: "Failed to open video device" with supported camera - device id mismatch

jl2008, 2019-04-10:
**Short description**: the camera device id shown in the admin interface does not match the file under `/dev/v4l/by-id/ `and the camera is therefore not recognized. Thanks in advance for looking into it.

The camera (link to the model further below, listed as compatible with MotionEyeOS and used by others) has an id seen below:

```
[root@meye-0a927c1d ~]# ls -al /dev/v4l/by-id/
total 0
lrwxrwxrwx    1 root     root            12 Dec 31  1969 usb-HD_Camera_Manufacturer_USB_2.0_Camera-video-index0 -> ../../video0

[root@meye-0a927c1d ~]# ls -al /dev/video0
crw-------    1 root     root       81,   0 Dec 31  1969 /dev/video0
```
However, the software tries a different id:
```
[1:ml1] [ALR] [VID] v4l2_start: Failed to open video device /dev/v4l/by-id/usb-046d_08b2-video-index0: No such file or directory
[1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
```

### motionEyeOS Version

I am running motionEyeOS version: **20190119** (installed fresh, not upgraded)

motionEye Version | 0.39.3
Motion Version | 4.1.1

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2 

My camera model is: https://www.amazon.com/gp/product/B07C2RL8PB

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none.

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[motion.log](https://github.com/ccrisan/motioneyeos/files/3066034/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3066035/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3066036/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3066038/dmesg.log)

jl2008, 2019-04-10:
Update: I was able to fix it by forcing the device id to be `/dev/video0 `in `thread-1.conf` and rebooting.  I'll leave this ticket open for a few days or so in case, Calin, you want to have a look at it; I'll close it afterwards in case you don't do it before.

ccrisan, 2019-04-12:
@jl2008 this looks like a valid issue so I'm not going to close it. Nevertheless, it's strange how this could happen. The piece of code that comes up with that path is at https://github.com/ccrisan/motioneye/blob/dev/motioneye/v4l2ctl.py#L210.

The only possible explanation I see at this time is the fact that somehow the persistent path has changed since the moment the camera was added. Can you reproduce the issue?

jl2008, 2019-04-13:
I'll get a second camera in a few weeks (this one is already installed and doing its job) and will retry. Odd, indeed. I did nothing special: installed MotionEyeOS and then plugged in the camera in the USB port. I think even rebooted when I saw the camera was not found. 

I've also just checked what happens if I plug the camera out and then back again, even changing the USB port, just in case it makes a difference. However, that old device name (`/dev/v4l/by-id/usb-046d_08b2-video-index0`) did not appear back, not even the logs (I grepped all in `/var/log` for the `usb-046d_08b2` substring). Powering off and on the Raspberry Pi doesn't make a difference either, the new name (`usb-HD_Camera_Manufacturer_USB_2.0_Camera-video-index0`) is the only one that appears.

jl2008, 2019-05-11:
An update: I got two more identical cameras. The issue was not present. I used the same version of motioneyeos. I can only shrug, so I'll close the bug.

closed 2019-05-11 by jl2008

MHPbvogt, 2020-06-18:
> Update: I was able to fix it by forcing the device id to be `/dev/video0 `in `thread-1.conf` and rebooting. I'll leave this ticket open for a few days or so in case, Calin, you want to have a look at it; I'll close it afterwards in case you don't do it before.

Thanks. Fixed it for me. I did change it /data/etc/camera-1.conf
`videodevice /dev/video0` had been set to video10 for some reason.

edumats, 2021-04-18:
My Raspberry Pi v2.1 camera was not working and looking at the logs, the following errors were present:

v4l2_device_capability: Device does not support capturing.
vid_start: V4L2 device failed to open

I suspect that it may have something to do with the fact that I inserted the camera flat cable with the wrong side up when I first booted my raspberry pi. Even when I switched the flat cable to the correct position it still displayed the same messages and I finally solved the issue by editing the camera-1.conf file and setting videodevice /dev/video0 instead of videodevice /dev/video10 as @Benjamin-Vogt suggested.

Working perfectly now and I hope this helps somebody with the same issue.

MitchDresdner, 2021-11-13:
Had same issue as reported by Benjamin above, June 18, 2020.  Resolved by changing from video10 to /dev/video0

motionEye Version0.42.1
Raspberry Pi mdl3 B

Phoenix1747, 2023-01-11:
Have the same issue and don't want to use /dev/videoX as it constantly moves once the device restarts or the USB gets plugged in and out.

pauldogg, 2023-12-10:
Same here. Device occasionally becomes /dev/video10, changing it back to /dev/video0 solves the problem.