motioneye-project/motioneyeos #1890: Web interface not working through wifi

Clubeddie, 2019-04-11:
First of all let me say that you build a very cool tool! I have tested it through cable and it is a very easy setup and great configuration options.

I only have an issue. I installed motioneyeos-raspberrypi3-dev20190408 on a RPI 3B+. Placed the ssh and wpa file on the boot partition. And booted it up.

I can ping the static Ip address of the raspberry (also checked the right IP adress in my router) But cannot connect the web interface via this IP.

When i put in a cable in the raspberry, the web interface is reachable and working through that IP adress (wired connection). I can ping both adresses. And the setup is correct, so the installation is succesfull.

So i tried a new setup, with a fresh install where no wired connection is used. But then also the web interface is not working. As soon as i connect the cable find the correct IP adress, web interface is reachable.

I only can connect with SSH on the IP address of the cable connection. When i check configuration with ifconfig/iwconfig. I see a wifi connection with correct data.

What am i doing wrong?

Clubeddie, 2019-04-11:
Strange...after doing nothing 2 hours later i can acces motioneyeos web interface via wifi?.... i will check the next couple of days if it is still keeps working

rotrials, 2019-04-12:
If you use a static ip for for RPI, check if there is another device with the same ip in your local network. This is the only thing which came into my mind.

jasaw, 2019-04-12:
@Clubeddie After device boot up, the web interface takes quite a long time to load. You might have to wait for 5 minutes after boot up and check again.
Another possibility is weak wifi signal. The fact that you can ping and ssh the Pi means basic connectivity is there. Opening the web interface requires a lot of data to be transferred, like the live video stream, which chokes when wifi signal is weak.

Clubeddie, 2019-04-14:
I don't know what it was. The Wi-Fi connection was good. Approx 1 meter between raspberry and accesspoint. I noticed that at booting and several minutes after there was a glithc. But it is working perfectly now.

closed 2019-04-14 by Clubeddie