motioneye-project/motioneyeos #1895: Standalone access point - first boot process hangs and reboots constantly

Ke-And, 2019-04-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board

I am using the following board/model: Raspberry PI 3B+).

### Camera = 

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: V4L2 = Pi Camera produced by Waveshare)

### Network Connection:  

Should work as standalone network, but there is a problem in the startup sequence

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 not applicable (yet).
____________________________________

Hello Calin

I am completely new to the Raspberry Pi and the working of Linux.
So possibly I am doing things wrong.

I started with installing MotionEye on my Pi in a network environment.
MotionEye works excellent, both with an UTP-cable and via Wifi on a home network connected to internet.
Then I disconnected my internet connection.
MotionEye still works perfectly via Wifi and is accessible with tablet, pc or smartphone.

But I need to have the PI functioning as a stand alone network / access point.
So I followed your description on this subject and created the hostapd.conf and dnsmasq.conf files exactly as you desrcibed.
And copied put them on a new SD-card with your MotionEye image, in the root of the card.

After start up of de Pi and the processing of the hostapd.conf file I get an error at the "Configuring network" row.
See the screen picture below.

Can you tell me what is wrong and how I can solve that?
Hope I don't ask something stupid.

Thanks in advance.
Best regards

André Kemper, the Netherlands.

[image]

ccrisan, 2019-04-16:
Set [OS_NETWORKLESS](https://github.com/ccrisan/thingos/wiki/os.conf#os_networkless) to `true` if you have no uplink.

Ke-And, 2019-04-16:
Hello Calin

Thanks for your quick reply.

Do I have to make a file os.conf  with only one statement OS_NETWORKLESS=true  -and  and put that in the root of de SD-card?

I understand that the PI commands are capital sensitive, so OS-NETWORKLESS should be all capital I guess.

Apart form the hostapd.conf and dnsmasq.conf files do I also need to put a file wpa-supplicant.conf  (with the SSID-name en passkey) and an empty shh.txt in the root?

Best regards

André Kemper

ccrisan, 2019-04-19:
Yes, you'll need to create the file yourself and add it to your boot partition. Yes, it's case-sensitive. The `ssh.txt` file is useless and won't do anything on motionEyeOS.

closed 2019-04-19 by ccrisan