motioneye-project/motioneyeos #1900: Recorded video bad quality

Kalasgurka, 2019-04-27:
I have an issue with the video quality, ive set my camera to both record and photo when there is motion.
Photos are great but the video is wierd, colors are all smeared and pixely. When i unplug and replug the raspberry, the first video it records of me is always great quality and no wierd color smears.
Im using the latest myeyeos, raspberry pi 3 model b+ and the standard raspberry camera.
Might also add im not used to python at all, thats why myeyeos is great for me.

How can i solve this?

Kalasgurka, 2019-04-30:
This is how it looks:
https://imgur.com/a/dOoFzkO

user9023523, 2019-04-30:
>When i unplug and replug the raspberry, the first video it records of me is always great quality 
Sounds like your cable or wall plug is inadequate. Check your logs and see if they mention undervoltage. I had to get a thicker gauge cable to prevent undervoltage.

jasaw, 2019-05-01:
@Kalasgurka Was that video encoded using H264/OMX? I suspect that particular video has very low bit rate. Can you please check the bit rate of this particular video?

I know there's a bug where H264/OMX encoded video sometimes produces very low bit rate. I haven't been able to reproduce it reliably, and haven't been able to identify which bit of software is responsible. If you manage to reliably reproduce it, please document it step by step.

rotrials, 2019-05-01:
I use the motioneyeos. Raspberry pi 2 and "Night vision camera module for Raspberry Pi"(https://shop.pimoroni.de/products/night-vision-camera-module-for-raspberry-pi?variant=12176175366246)

I have the same issue, the Frame rate is: 21fps. The wall plug is the original one in my case.

Lookslikerain75, 2019-05-05:
I am having the same issue H264/OMX saves really poor quality mp4 files. I have tried different USB and SD cards not difference. It was working fine, standard H264 works perfectly. If there are any entries in the log files i need to look for let me know. using H264 for now.

rotrials, 2019-05-05:
I have 2 more raspberry pi and different cams, and the quality is ok.  Please have a look at the picture provided by @jasaw [https://imgur.com/a/dOoFzkO](url) it looks like a video codec problem.

coderpop, 2019-05-11:
I had same problem - the solution that worked for me was to reduce the video quality to 90%.

Give it a try and see if that helps you.

rotrials, 2019-05-11:
@coderpop I have 75% for streaming and movie quality :(

jasaw, 2019-05-12:
Which motionEyeOS version are you guys running? Let's see if we can narrow it down. I am seeing this issue myself, but can't reliably reproduce the problem.

rotrials, 2019-05-13:
I'm using 0.40rc5. Raspberry PI2 (with original power supply) and MMAL camera. It's hard to reproduce :( this why I didn't created an issue. I have tried to rewrite the sd card several times (on 2 different sd cards, sandisk ultra and extreme), but with no success, the problem still persist.

Lookslikerain75, 2019-05-13:
I am using version  20190427 on a PI 3, issue started after upgrading to this version and i believe an under powered power supply. Not sure if the under volting caused the issue but switched to a better power supply, no under volting now, but issues but still had poor video quality capture using OMX. The only way I could get the video quality fixed was to set the video resolution to the lowest setting apply then change back to the higher resolution, clicked Apply video recording has been fine ever since 5 days running no issues with restarts and power offs. If this the under volting was the issue it seems something in the config did not revert back to a higher quality after power issue restored, forcing resolution change fixed that I assume by forcing change in the config. However I have not been able to replicate this to prove this is a consistent root cause and resolution. Just my thoughts probably way off or a coincidence it now works.

rotrials, 2019-05-20:
I now have 3 RPIs:
1. Raspberry pi 3B+ with CSI 5-megapixel camera .
2. Raspberry pi 3 with CSI 5-megapixel camera .
3. Raspberry pi 2 with CSI 8-megapixel camera .
All with the latest motioneyeos and one odroid xu4 (arch linux + motioneye) which is the main server.
All 3 RPIs records sometimes videos with the bad quality. But even when the recorded video is bad, the streaming is ok.
What I also saw is the "data rate" and "total bitrate". For a bad quality is around 134kbps and for well recorded video is around 8592kbps.
A week before I had two of the RPIs with a usb cam and the recorded video was always fine. 
I hope the details above will help and somebody will take his time to solve this issue :( It's very frustrating because I use this system for security and I'm struggling with it for long time :( .
I would avoid to open another issue in the right manner (with logs) but which will be the same like this one. Thanks

rotrials, 2019-06-06:
One more observation, If I reduce the resolution it seems that it's able to record correctly the videos:
For Raspberry pi3B+ the resolution is 1280x720
For Raspberry pi2 the resolution is 1280x800
Before was higher than 1280x1xxx

Somebody any ideas? I don't use heatsinks :( I think I should try with a heatsink...

jasaw, 2019-06-07:
I've looked through motion source code, but can't see anything obviously wrong with it. I have reported the issue to the Raspberry Pi guys: https://github.com/raspberrypi/firmware/issues/1151

In the meantime, if you guys want to debug further, you could try turning on debug and look for this line in motion log file when you get low bit rate:
> codec vbr/crf/bit_rate: [video_quality]

This line tells us the actual video quality parameter given to ffmpeg.

rotrials, 2019-06-17:
@jasaw 
Does this help? 

**1.Bit Rate**
[image]

**2.codec vbr/crf/bit_rate: [video_quality]**
I have enabled the the "Enable Debugging" from "Expert settings". I checked: motion.log, motioneye.log, messages.log, boot.log and dmesg.log but I did not find the video quality :( But I found the following:

Jun 17 20:37:59 Cam-Terasa user.info motion: [1:ml1:Cam-Terasa] [INF] [ENC] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: **4000**

Jun 17 20:07:56 Cam-Terasa user.info motion: [1:ml1:Cam-Terasa] [INF] [ENC] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: **3375000**

Is this bit_rate difference normal?
Am I looking in the right log file?
Should I try something else?
Thank you for your interest in this topic

jasaw, 2019-06-18:
@rotrials That's exactly what I need. Thank you for testing this. This looks like a motion software bug. Motion obviously set the video quality to 4kbps.
> Jun 17 20:37:59 Cam-Terasa user.info motion: [1:ml1:Cam-Terasa] [INF] [ENC] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 4000

I've report this to motion developers: https://github.com/Motion-Project/motion/issues/958

rotrials, 2019-06-18:
Glad to hear that I could help and we are moving forward with this topic :) 
This issue is very frustrating. I hope somebody will have time to fix it.
[motion.log](https://github.com/ccrisan/motioneyeos/files/3303235/motion.log)

jasaw, 2019-06-21:
Hi guys,

I've put together a potential fix for this problem here: https://github.com/jasaw/motioneyeos/releases/tag/fix-bitrate-dev20190619

You guys are free to give it a whirl, but keep in mind that this is not fully tested. Don't load it on system critical cameras.

rotrials, 2019-06-23:
@jasaw I have checked it with rpi 3B+ and for 9 movies recorder at 1600x1200 all are ok. But I will keep testing it in the following days. I will let you know what is the status :) . 
I have linked the camera to my odroid server (archlinux) and when I'm trying there to change some settings I received "An error occurred. Refreshing is recommended." do you know why? When I connect directly to the RPI there was no problem. Thanks you.

jasaw, 2019-06-24:
@rotrials I don't know why you get "An error occurred. Refreshing is recommended.". Maybe it has something to do with browser cache? Try clearing browser cache?

If this fixes the low bitrate problem, we will roll it into official motionEyeOS as a patch until it's merged into upstream motion.

rotrials, 2019-06-24:
@jasaw strange, it's still not working. I have tried on mobile phone and 2 browsers on the PC. 
And I cannot test it for at least 2 days, because I have to change some settings when i'm outside the local network :(

jasaw, 2019-06-26:
@rotrials I didn't change anything except patched the motion software in this test image. I don't know why it's giving you an error on the web browser. @ccrisan You're the expert on motionEye, any thoughts?

ccrisan, 2019-06-26:
Guys, this usually happens if the JavaScript code is out of date and doesn't match with what the server expects. That's fixed by clearing the browser's cache, as @jasaw indicated. If that still doesn't fix the problem, we need to see the motioneye.log file.

rotrials, 2019-06-26:
@ccrisan  I have 3 RPI with motioneyeos. And 1 odroid xu4 (archlinux and motioneye).
One of the raspberry pi 3B+ is with the motioneyeos from jasaw. When I access the web interface of the RPI, I can change anything with no errors. But when I access it from the odroid "server" I always get that error.

Below is a part of the log:

2019-06-26 21:59:09: [motioneye]     INFO: hello! this is motionEye server 0.40
2019-06-26 21:59:12: [motioneye]     INFO: cleanup started
2019-06-26 21:59:12: [motioneye]     INFO: wsswitch started
2019-06-26 21:59:12: [motioneye]     INFO: tasks started
2019-06-26 21:59:12: [motioneye]     INFO: mjpg client garbage collector started
2019-06-26 21:59:12: [motioneye]     INFO: smb mounts started
2019-06-26 21:59:12: [motioneye]     INFO: server started
2019-06-26 22:00:33: [motioneye]    ERROR: 'max_frame_change_threshold'
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 290, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 530, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 369, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 762, in motion_camera_ui_to_dict
KeyError: 'max_frame_change_threshold'
2019-06-26 22:00:33: [motioneye]    ERROR: 500 POST /config/1/set/?_username=admin&_admin=true&_signature=2db401a6e42eb659d67e80ecee4e8d846112ddef (192.168.1.11) 19.11ms

Does this help? 
192.168.1.11 is the odroid

Thank you

ccrisan, 2019-06-26:
@rotrials clear your browser's cache, refresh and retry.

rotrials, 2019-06-26:
@ccrisan I have done this, I tried on one phone and 2 PCs. This is what log says?

ccrisan, 2019-06-26:
Log says that your UI does not provide the new `max_frame_change_threshold` option, which usually happens due to browser caching. It may not be the only cause though :)

rotrials, 2019-07-03:
I have tried write latest motioneyeos img. (motioneyeos-raspberrypi3-20190619.img.xz) to see if the problem is from jasaw img or from my odroid xu4 (arch linux).
2 times works fine, and I get the same error with the original motioneye img.
It seems that the problem is not from jasaw img. I assume that this is due to some updates from the arch linux :(
I don't know what else I can try or how I can solve this.

jasaw, 2019-07-05:
@rotrials Did my test image fix your low quality recording problem?

rotrials, 2019-07-05:
@jasaw  I will summarize : I tested your image for a couple of minutes and it seems that it works no frame rate error, to be sure I wanted to check it for at least 1 day and this is not possible because of "An error occurred. Refreshing is recommended." error which is not due to the browser cache.

ilovelinux, 2019-07-12:
Same issue here, even with path. Here's my log:
```
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] motion_startup: Using log type (ALL) log level (ALL)
[0:motion] [INF] [ALL] [Jul 12 12:52:19] conf_output_parms: Writing configuration parameters from all files (2):
[0:motion] [INF] [ALL] [Jul 12 12:52:19] Thread 0 - Config file: /usr/local/etc/motion/motion.conf
[...]
[0:motion] [INF] [ALL] [Jul 12 12:52:19] camera_dir                /etc/motion/conf.d
[0:motion] [INF] [ALL] [Jul 12 12:52:19] Thread 1 - Config file: /etc/motion/conf.d/picamera.conf
[0:motion] [INF] [ALL] [Jul 12 12:52:19] mmalcam_name              vc.ril.camera
[0:motion] [INF] [ALL] [Jul 12 12:52:19] width                     640
[0:motion] [INF] [ALL] [Jul 12 12:52:19] height                    480
[0:motion] [INF] [ALL] [Jul 12 12:52:19] framerate                 30
[0:motion] [INF] [ALL] [Jul 12 12:52:19] movie_output              on
[0:motion] [INF] [ALL] [Jul 12 12:52:19] movie_max_time            0
[0:motion] [INF] [ALL] [Jul 12 12:52:19] movie_quality             100
[0:motion] [INF] [ALL] [Jul 12 12:52:19] movie_codec               mp4:h264_omx
[0:motion] [INF] [ALL] [Jul 12 12:52:19] movie_filename            %Y-%m-%d/%H-%M-%S
[0:motion] [DBG] [ALL] [Jul 12 12:52:19] motion_ntc: v4l2   : available
[0:motion] [DBG] [ALL] [Jul 12 12:52:19] motion_ntc: bktr   : not available
[0:motion] [DBG] [ALL] [Jul 12 12:52:19] motion_ntc: webp   : available
[0:motion] [DBG] [ALL] [Jul 12 12:52:19] motion_ntc: mmal   : available
[0:motion] [DBG] [ALL] [Jul 12 12:52:19] motion_ntc: ffmpeg : available
[0:motion] [DBG] [DBL] [Jul 12 12:52:19] motion_ntc: mysql  : not available
[0:motion] [DBG] [DBL] [Jul 12 12:52:19] motion_ntc: sqlite3: not available
[0:motion] [DBG] [DBL] [Jul 12 12:52:19] motion_ntc: pgsql  : not available
[0:motion] [DBG] [DBL] [Jul 12 12:52:19] motion_ntc: nls    : available
[...]
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] motion_start_thread: Camera ID: 1 is from /etc/motion/conf.d/picamera.conf
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] motion_start_thread: Camera ID: 1 Camera Name: (null) Device: /dev/video0
[0:motion] [NTC] [ALL] [Jul 12 12:52:19] main: Waiting for threads to finish, pid: 1597
[1:ml1] [NTC] [ALL] [Jul 12 12:52:19] motion_init: Camera 1 started: motion detection Enabled
[1:ml1] [NTC] [VID] [Jul 12 12:52:19] vid_start: Opening MMAL cam
[1:ml1] [NTC] [VID] [Jul 12 12:52:19] mmalcam_start: MMAL Camera thread starting... for camera (vc.ril.camera) of 640 x 480 at 30 fps
[1:ml1] [NTC] [VID] [Jul 12 12:52:19] create_camera_component: MMAL camera component created
[1:ml1] [NTC] [ALL] [Jul 12 12:52:19] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [NTC] [ALL] [Jul 12 12:52:19] image_ring_resize: Resizing pre_capture buffer to 25 items
[1:ml1] [INF] [EVT] [Jul 12 12:52:22] event_new_video: Source FPS 27
[1:ml1] [NTC] [ENC] [Jul 12 12:52:22] ffmpeg_set_codec: Using codec h264_omx
[1:ml1] [INF] [ENC] [Jul 12 12:52:22] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 5832000
[1:ml1] [NTC] [EVT] [Jul 12 12:52:22] event_newfile: File of type 8 saved to: ./2019-07-12/12-52-21.mp4
[1:ml1] [NTC] [ALL] [Jul 12 12:52:22] motion_detected: Motion detected - starting event 1
[1:ml1] [DBG] [ENC] [Jul 12 12:52:22] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:52:22] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] [Jul 12 12:52:22] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:52:22] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] [Jul 12 12:52:23] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:52:23] ffmpeg_put_image: Buffered packet
[1:ml1] [NTC] [ALL] [Jul 12 12:53:00] mlp_actions: End of event 1
[1:ml1] [INF] [EVT] [Jul 12 12:53:03] event_new_video: Source FPS 29
[1:ml1] [NTC] [ENC] [Jul 12 12:53:03] ffmpeg_set_codec: Using codec h264_omx
[1:ml1] [INF] [ENC] [Jul 12 12:53:03] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 6264000
[1:ml1] [NTC] [EVT] [Jul 12 12:53:03] event_newfile: File of type 8 saved to: ./2019-07-12/12-53-02.mp4
[1:ml1] [NTC] [ALL] [Jul 12 12:53:03] motion_detected: Motion detected - starting event 2
[1:ml1] [DBG] [ENC] [Jul 12 12:53:03] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:53:03] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] [Jul 12 12:53:03] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:53:03] ffmpeg_put_image: Buffered packet
[1:ml1] [NTC] [ALL] [Jul 12 12:53:25] mlp_actions: End of event 2
[1:ml1] [INF] [EVT] [Jul 12 12:53:53] event_new_video: Source FPS 28
[1:ml1] [NTC] [ENC] [Jul 12 12:53:53] ffmpeg_set_codec: Using codec h264_omx
[1:ml1] [INF] [ENC] [Jul 12 12:53:53] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 6048000
[1:ml1] [NTC] [EVT] [Jul 12 12:53:53] event_newfile: File of type 8 saved to: ./2019-07-12/12-53-52.mp4
[1:ml1] [NTC] [ALL] [Jul 12 12:53:53] motion_detected: Motion detected - starting event 3
[... cut ...] Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:53:53] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] [Jul 12 12:53:53] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] [Jul 12 12:53:53] ffmpeg_put_image: Buffered packet
[1:ml1] [NTC] [ALL] [Jul 12 12:53:59] mlp_actions: End of event 3
[...]
```
Even with that bit rate, actual output video bit rate is about 2000kb/s and video quality is still a garbage

rotrials, 2019-07-12:
@ilovelinux 
Please try the fix from jasaw.

> Hi guys,
> 
> I've put together a potential fix for this problem here: https://github.com/jasaw/motioneyeos/releases/tag/fix-bitrate-dev20190619
> 
> You guys are free to give it a whirl, but keep in mind that this is not fully tested. Don't load it on system critical cameras.

ilovelinux, 2019-07-12:
As I wrote motion is running with the path applied:
```
$ git diff
diff --git a/ffmpeg.c b/ffmpeg.c
index a8afa6e..3dee026 100644
--- a/ffmpeg.c
+++ b/ffmpeg.c
@@ -558,7 +558,7 @@ static int ffmpeg_set_quality(struct ffmpeg *ffmpeg){
         if ((strcmp(ffmpeg->codec->name, "h264_omx") == 0) || (strcmp(ffmpeg->codec->name, "mpeg4_omx") == 0)) {
             // H264 OMX encoder quality can only be controlled via bit_rate
             // bit_rate = ffmpeg->width * ffmpeg->height * ffmpeg->fps * quality_factor
-            ffmpeg->quality = (ffmpeg->width * ffmpeg->height * ffmpeg->fps * ffmpeg->quality) >> 7;
+            ffmpeg->quality = (int)(((int64_t)ffmpeg->width * ffmpeg->height * ffmpeg->fps * ffmpeg->quality) >> 7);
             // Clip bit rate to min
             if (ffmpeg->quality < 4000) // magic number
                 ffmpeg->quality = 4000;
@@ -595,7 +595,7 @@ static int ffmpeg_codec_is_blacklisted(const char *codec_name){
          * - remove the "h264_omx" from this blacklist.
          * More information: https://github.com/Motion-Project/motion/issues/433
          */
-        "h264_omx",
+//        "h264_omx",
     };
     size_t i;
 
```

ilovelinux, 2019-07-12:
I thought to have applied changes but I did not. I built motion again and now it works. I'll test it and I'll let you know if there are any problems

haha150, 2019-07-19:
> 
> 
> Hi guys,
> 
> I've put together a potential fix for this problem here: https://github.com/jasaw/motioneyeos/releases/tag/fix-bitrate-dev20190619
> 
> You guys are free to give it a whirl, but keep in mind that this is not fully tested. Don't load it on system critical cameras.

is this getting merged to master?

jasaw, 2019-07-22:
Created PR for this issue: https://github.com/ccrisan/motioneyeos/pull/2006

@haha150 I'm not 100% sure that this patch will fix the bad quality issue, and I was hoping to get some feedback from users.

jasaw, 2019-08-29:
@haha150 @ilovelinux @rotrials @Kalasgurka Do you guys still see the low bit rate issue?

haha150, 2019-08-29:
Not when running the nightly build and lowering the quality

jasaw, 2019-09-09:
Should be fixed now.

closed 2019-09-09 by jasaw