motioneye-project/motioneyeos #1902: Pi3 migrate to Pi Zero

egrp, 2019-05-01:
I have installed MotionEye on a Pi3, using the latest Jessie and MoionEye software (Not MotioEyeOs as i could not get my scavenged laptop camera to work with MotioEyeOS).
Everything works on a Pi3, i tried to move everything to a Pi ZeroW and I get no video feed.
Do i need a different version to install on the Pi Zero? (I have not yet been able to get this camera to work).

I did have 3 Pi Zero W's with actual USB cameras that i took on vacation and everything worked great.
I am trying to make the package slimmer, and i have a few laptop cameras on hand. 
I was able to get everything working on the Pi3, but using the same install methods, i get no video stream on the Pi Zero, 
I know this has to be something i have missed, but i am not able to get this to work.

Any advice / direction would be most appreciated.

Thank you in advance

jasaw, 2019-05-10:
Pi Zero W uses different image from Pi3 .
You need [this](https://github.com/ccrisan/motioneyeos/releases/download/20190427/motioneyeos-raspberrypi-20190427.img.xz) image for Pi Zero W.

egrp, 2019-05-10:
Thank you for the response..
I have the Pi Zero image up and running using a Microsoft Lifecam as well as a Logitec Webcam, these work great.
I would like to get a reclaimed Toshiba Laptop Webcam to work on this as well, when I connect it, i do not get any video feed from it,
I compiled MotionEye from source code onto a Raspberry Pi 3 and this camera does work after following others instructions. I have not been able to make this webcam work on the Pi zero following the same instructions and not sure what I am doing wrong.

jasaw, 2019-05-13:
@egrp This sounds like a [motionEye](https://github.com/ccrisan/motioneye) issue, not motionEye**OS**. You might get better response if you post your issue there.

ccrisan, 2019-07-15:
Please make sure the camera works well with Motion first. If it does, it will work with motionEye as well.

closed 2019-07-15 by ccrisan