motioneye-project/motioneyeos #1903: MJPEG URLS in Version 20190427

helmar74, 2019-05-02:
Hi,

I only have a question.
I upgraded my cams to 20190427 successfully. 
In the Changelog i read:
"motionEye: added support for mjpeg://" 

Does this mean, I can access the stream of the camera with a mjpeg URL?
I tried different URLs, but I don't manage to get it running...
Which port do I have to use?
Thanks a lot

helmar

Nypan-sr, 2019-05-04:
IP:8081

helmar74, 2019-05-05:
it's the "normal" streaming port?
If I change it here:
[image]

Than I have to use that one?

ccrisan, 2019-05-19:
> Does this mean, I can access the stream of the camera with a mjpeg URL?

It doesn't mean that, but you could have done that before, assuming your MJPEG client accepted the `mjpeg://` scheme.

This is about motionEye being able to pass a `mjpeg://` netcam URL to the underlying Motion daemon.

> Which port do I have to use?

It's `8081` unless you changed it explicitly.

closed 2019-05-19 by ccrisan