motioneye-project/motioneyeos #1905: More then 3 Cams are not possible?

vplustute, 2019-05-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190413

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: 4 Network Camera and 1 Remote MotionEye

My camera model is:
1x IPCAM (C9F0SgZ0N0P4L0)
2x IPCAM (C9F0SeZ0N0P7L0)
1x Grandstream GXV3662FHD
1x Remote MotionEye at Zero PI WH

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
no other peripherals

### Log Files

I consider the following log files relevant to this issue:

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3140722/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3140721/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3140724/motion.log)
[image]
[image]

When I try to add the fourth ipcam, I type in the adress of the cam, username und Password and the Motioneye detect the cam. But when I confirm my Settings I see the same stream like Camera ID 1. 
What can I do, that I can connecet all my Cams to Motion Eye. At an older Version it was working.

ccrisan, 2019-05-03:
The "Camera Device" field indicates different URLs and I assume those are the correct ones. If you individually add  Camera 1 and Camera 3, do they show their correct stream?

vplustute, 2019-05-06:
The URLs at  "Camera Device" field are correct.

I deleted all the cameras. Then I add all my cameras in a different order. At the Moment everything is working but I don't know why.

closed 2019-06-18 by vplustute