motioneye-project/motioneyeos #1906: Rollback from 20190427

duckietm, 2019-05-03:
I had to rollback to version 20190119, on both raspberry's I get camera drops :

May  3 10:36:56 meye-ee94a9f9 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] Above message repeats 1 times
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  3 10:37:04 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May  3 10:37:29 meye-ee94a9f9 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] Above message repeats 1 times
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started

2.986139] input: Microsoft\xc2\xae LifeCam HD-3000: Mi as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input0
[  537.696286] uvcvideo: Failed to resubmit video URB (-1).
[ 1893.630426] uvcvideo: Non-zero status (-5) in status completion handler.
[ 1896.810347] uvcvideo: Failed to resubmit video URB (-1).
[ 1929.968553] uvcvideo: Failed to resubmit video URB (-1).
[ 3388.523463] uvcvideo: Failed to resubmit video URB (-1).
[ 4150.957259] uvcvideo: Failed to resubmit video URB (-1).
[ 4747.644293] uvcvideo: Failed to resubmit video URB (-1).
[ 5012.836327] uvcvideo: Failed to resubmit video URB (-1).

May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  3 10:37:36 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May  3 10:38:02 meye-ee94a9f9 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] Above message repeats 1 times
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  3 10:38:08 meye-ee94a9f9 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)

duckietm, 2019-05-03:
2019-05-03 09:37:09: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:10: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:10: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:10: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:10: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:10: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:11: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2019-05-03 09:37:12: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED

duckietm, 2019-05-03:
After clearing the camera /data/etc/camera-1.conf and rebooting I get the following:

duckietm, 2019-05-03:
in the motion.conf :

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] Above message repeats 1 times
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[2:ml2:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[2:ml2:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[2:ml2:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[2:ml2:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0: No such file or directory
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
cccuuurrrlll:::   (((555666)))   RRReeecccvvv   fffaaaiiillluuurrreee:::   CCCooonnnnnneeeccctttiiiooonnn   rrreeessseeettt   bbbyyy   pppeeeeeerrr

curl: (56) Recv failure: Connection reset by peer

duckietm, 2019-05-03:
After some digging I found the solution, in /data/etc/camera-1.conf I have changed the videodevice to

videodevice /dev/video0

This was a problem before : https://github.com/ccrisan/motioneye/issues/469

duckietm, 2019-05-03:
After the above Fixed, here is the next problem 

May  3 12:18:54 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  3 12:18:57 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May  3 12:19:27 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May  3 12:19:29 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)

2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:46: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:47: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:47: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:47: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2019-05-03 12:21:47: [motioneye]    ERROR: motion not running, starting it

The Camera reloads every 40 sec. ???

duckietm, 2019-05-03:
There is something realy wrong with this release :(

ccrisan, 2019-05-03:
Rolling back to a pre-20190427 version won't go smooth because file formats have been changed (we're now using Motion 4.2 with their new config options).

We provided a script to automatically convert config files from older Motion versions, a script that's being executed upon upgrade. However there's no automatic conversion of config files from 4.2 to older versions so that's why neither rolling back nor backup/restore work.

At this point I recommend that you rewrite your OS image from scratch or at least remove all cameras and re-add them while running 20190427.

duckietm, 2019-05-03:
Thanks, I will give that a go.

duckietm, 2019-05-03:
p.s.  remove all cameras  will not work see above :

[root@meye-ee94a9f9 by-id]# ls
 usb-Microsoft_Microsoft??_LifeCam_HD-3000-video-index0@
 [root@meye-ee94a9f9 by-id]# ls -ltr
 total 0
 lrwxrwxrwx 1 root root 12 Jan 1 1970 usb-Microsoft_Microsoft??_LifeCam_HD-3000-video-index0 -> ../../video0

Hmmm Microsoft_Microsoft??_LifeCam

closed 2019-05-07 by duckietm