motioneye-project/motioneyeos #1907: New install 20190427

duckietm, 2019-05-03:
Hello,

I did an clean install of version : 20190427. (have done this on 4 devices)
But still a lot of problems, Camera keeps reseting.
I have added the logs
[boot.log](https://github.com/ccrisan/motioneyeos/files/3142823/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3142824/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3142825/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3142826/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3142827/motioneye.log)

If i install version 20190119  all works perfectly, so it realy looks like there is an problem.

duckietm, 2019-05-03:
motioneye.log from version 20190119:

2019-05-03 16:59:42: [motioneye]     INFO: hello! this is motionEye server 0.39.3
2019-05-03 16:59:43: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2019-05-03 16:59:43: [motioneye]     INFO: cleanup started
2019-05-03 16:59:43: [motioneye]     INFO: wsswitch started
2019-05-03 16:59:43: [motioneye]     INFO: tasks started
2019-05-03 16:59:43: [motioneye]     INFO: mjpg client garbage collector started
2019-05-03 16:59:43: [motioneye]     INFO: smb mounts started
2019-05-03 16:59:43: [motioneye]     INFO: server started
2019-05-03 17:00:57: [motioneye]     INFO: adding new v4l2 camera with id 1...
2019-05-03 17:00:57: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2019-05-03 17:02:20: [motioneye]     INFO: rebooting
2019-05-03 17:02:20: [motioneye]     INFO: executing "/sbin/reboot"
2019-05-03 17:02:21: [motioneye]     INFO: interrupt signal received, shutting down...
2019-05-03 17:02:21: [motioneye]     INFO: server stopped
2019-05-03 17:02:21: [motioneye]     INFO: tasks stopped
2019-05-03 17:02:23: [motioneye]     INFO: motion stopped
2019-05-03 17:02:23: [motioneye]     INFO: smb mounts stopped
2019-05-03 17:02:23: [motioneye]     INFO: bye!
2019-05-03 17:02:44: [motioneye]     INFO: hello! this is motionEye server 0.39.3
2019-05-03 17:02:47: [motioneye]     INFO: cleanup started
2019-05-03 17:02:47: [motioneye]     INFO: wsswitch started
2019-05-03 17:02:47: [motioneye]     INFO: tasks started
2019-05-03 17:02:47: [motioneye]     INFO: mjpg client garbage collector started
2019-05-03 17:02:47: [motioneye]     INFO: smb mounts started
2019-05-03 17:02:47: [motioneye]     INFO: server started

And no more errors / logs after this, and camera's are fully stable.

Devices : Raspberry Pi 3B+ with M$ 3000HD camera's
The Camera's are 14 days old and the Rasberry are 4 days old all 4

Jese44, 2019-05-04:
Hey, i also updated the version to latest (20190427) and i'm unable to add rpi camera module as the camera that i'd use. Everything was fine with the previous version(s).

duckietm, 2019-05-04:
I have tested the following, on a new image of 20190427 :

- Raspberry 3B+ With Pi camera
- Raspberry 3B+ With 3000HD Microsoft camera
- Raspberry 3B+ With Logitech C920 HD
- Raspberry 3B+ With Trust GXT 1170

All with the same result :

WARNING: Connect error on fd 15: ECONNREFUSED
ERROR: motion not running, starting it

Also tested this on 20190427 all runs fine no errors what so ever
I also notice that the bootup time is taking much longer on 20190427 then on 20190119 

So I advise not to update to 20190427 and stay on 20190119 as this has no problems what so all

ccrisan, 2019-05-04:
@duckietm could you please enable debugging (in Expert Settings) and try to reproduce the problem with the Pi Camera only? Then attach your logs again in hope that we'll find the problem.

duckietm, 2019-05-05:
I have attached the log files from the M$ 3000HD in Debug mode as the current one with the Pi camera takes me 1 hour to get it out of the camera :(

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3145218/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3145219/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3145220/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3145221/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3145222/motion.log)

duckietm, 2019-05-07:
Any luck looking into the logs?

ccrisan, 2019-05-07:
There's a weird error in `messages.log`:

    ...
    May  5 07:04:46 meye-11da7c58 user.debug motion: [1:ml1:Camera1] [DBG] [VID] v4l2_capture: 1) vid_source->pframe 2
    May  5 07:04:46 meye-11da7c58 user.debug motion: [1:ml1:Camera1] [DBG] [VID] v4l2_capture: 2) vid_source->pframe 2
    May  5 07:04:46 meye-11da7c58 user.debug motion: [1:ml1:Camera1] [DBG] [VID] v4l2_capture: 3) vid_source->pframe 3 vid_source->buf.index 3
    May  5 07:04:46 meye-11da7c58 user.debug motion: [1:ml1:Camera1] [DBG] [VID] v4l2_capture: the_buffer index 3 Address (72f44000)
    May  5 07:04:46 meye-11da7c58 user.debug motion: [0:cn0] [DBG] [STR] webu_parseurl: Sent url: HTTP/1.0
    May  5 07:04:46 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
    May  5 07:04:46 meye-11da7c58 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (ALL)
   ...

Somehow it gets an invalid URL (`Invalid url: HTTP/1.0`) and then restarts. @duckietm is this a clean install with no extra scripts? If yes, can I have your configuration files (`/data/etc/motion.conf` and `/data/etc/thread-1.conf`), after commenting out sensitive data from them?

duckietm, 2019-05-07:
Hello ccrisan,

Yes this a clean install, no scripts nothing. From scratch
I also tested the version dev20190413 / 20190413, they have the same problem.

duckietm, 2019-05-07:
I have attached motion.conf and the motioneye.conf (thread-1.conf has been replaced in the new build by motioneye.conf

closed 2019-05-07 by duckietm

duckietm, 2019-05-07:
indeed a lot of those

May  7 10:04:52 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:05:04 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:05:17 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:05:31 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:05:42 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:05:55 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:06:08 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:06:20 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:06:33 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:06:46 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:06:58 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:07:11 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:07:24 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:07:35 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:07:48 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:08:17 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:08:52 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:09:05 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:09:18 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:09:32 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:09:45 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:09:58 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:10:12 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:10:25 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:10:38 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:10:49 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:11:02 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:11:15 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.0
May  7 10:11:28 meye-11da7c58 user.err motion: [0:cn0] [ERR] [STR] webu_url_decode: Invalid url: HTTP/1.

duckietm, 2019-05-07:
I found the case of the problem !
As I am using an NAS (QNAP) to record the streams I found that after build 20190119 I have to add an / in the stream :(

[image]