motioneye-project/motioneyeos #1910: Privacy mask

RobiOnPi, 2019-05-06:
Hi,

Motion with MotioneyeOSis very useful camera home surveillance system.
**I think that one important setup is missing - privacy mask!**

Motion supports privacy mask configuration option - "mask_privacy" (https://motion-project.github.io/motion_config.html#mask_privacy). 

Example how to configure in MotioneyeOS:
- create masking file mask.pgm
- copying mask.pgm to ftp folder sdcard
- MotioneyeOS - Video Device - Extra Motion Option enter configuration option like this
  mask_privacy /home/ftp/sdcard/mask.pgm
And will work! Masked area on camera picture will be black!

**I think that is not difficult to add this new  setup option in MotioneyeOS for Privaacy Mask.** Mask editor is already made for motion mask. It can be used for creating new mask file for privacy. And enable option. When **Privacy Mask** is enabled new configuration option like mask_privacy /home/ftp/sdcard/mask.pgm must be added to camera setup!

**This will be a very useful add-on to MotioneyeOS.**

Thanks!

RobiOnPi, 2019-05-16:
Hi,
is this really so difficult to implement?
Privacy mask is really important because of GDPR (General Data Protection Regulation)!

lxmatic, 2019-05-30:
Hi, RobiOnPi you have prepared a great solution/example for MotioneyeOS and i believe @ccrisan will not miss that. I hope it will at least give some feedback about the possibility to include this useful option into MotioneyeOS.

BR,
Matic

jasaw, 2019-05-31:
This sounds like a [motioneye](https://github.com/ccrisan/motioneye) feature rather than motioneye**OS**. Can you please open this issue in motioneye project instead?

RobiOnPi, 2019-05-31:
OK, I understand! motioneyeOS is compilation of motion and motioneye.
Thank you
RobiOnPi

closed 2019-05-31 by RobiOnPi