motioneye-project/motioneyeos #1912: Motion Detection Not Working

DiversSundries, 2019-05-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
YES

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
YES
### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry PI CSI v2.1
### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals
None

### Log Files

I consider the following log files relevant to this issue:

 * motioneye.log
 * motion.log
 * motioneye.conf (renamed to .txt to upload)
 * thread-1.txt (renamed to .txt to upload)
 * example.jpg
[image]

[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/3155005/thread-1.txt)
[motioneye.txt]
(https://github.com/ccrisan/motioneyeos/files/3155004/motioneye.txt)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3154995/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3154996/motioneye.log)

DiversSundries, 2019-05-08:
Yikes! I should be careful what I wish for. I had missed **"Capture Mode = Motion Triggered"** under the Still Images section. After setting that, with the Minimum Motion Frames set to 1, Despeckle and Noise  Filters Off and Noise Level set to 0, the triggered images started pouring in. 

I'll play with these settings and try again. If all goes well, I'll see how it works under the latest version.
Perhaps this post will be useful to someone else.

ilovelinux, 2019-06-18:
I guess that this issue can be closed

closed 2019-06-18 by ccrisan