motioneye-project/motioneyeos #1913: First boot (sort of) works, but ceases to function on reboot.

ByteSizeOtter, 2019-05-08:
The very first boot of the system will work on a Raspberry Pi (Zero W or 3) and the cameras will function, however it will not allow me to change any settings, only producing an error on the page that something went wrong and refreshing is recommended when attempting to apply changes.

Attempts to reboot the system either via the web interface or SSH commands results in the device never coming back online. I have tried different types of SD cards and of course "turned it off and back on again", and tried the whole process with no cameras or peripherals attached to no avail. This seems like it could be similar to #1892 though this is occurring on Raspberry Pi.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi Zero W and Raspberry Pi 3.

### Camera

I am using the following type of camera: Raspberry Pi camera and/or Logitech C270 webcam.

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi or Ethernet (both tested)

ccrisan, 2019-05-19:
@ByteSizeOtter we need to see some logs. Please attach all of your logs, especially `motioneye.log` here.

ccrisan, 2019-07-15:
Closing due to no follow-up from OP.

closed 2019-07-15 by ccrisan