motioneye-project/motioneyeos #1914: SSH Remote Commands stop working on 20190427

kiwi-cam, 2019-05-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL and one USB

My camera model is: N/A

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
N/A

### Log Files

I consider the following log files relevant to this issue:
N/A (Logs only show a successful SSH connection, then disconnect)

############
I run a second Raspberry Pi running HomeBridge to enable/disable our indoor camera depending on whether anyone is (should be) home. This is achieving with shell scripts with remote ssh commands.

Up to 20190427 this command was run on the HomeBridge RPi:
`ssh -i /path/to/cert -o StrictHostKeyChecking=no admin@motioneye '/data/etc/enablecamera.sh'`

Running this command after upgrading to 20190427 resulted in output of
```
sh: cp: No such file or directory
sh: cp: No such file or directory
sh: grep: No such file or directory
sh: awk: No such file or directory
```
I believe this is because the non-interactive ssh session on the motioneyeos RPi no longer has a valid PATH variable so cannot find the executables for these rudimentary commands - this is just my working theory at this stage.

I've been able to workaround this by amending the command on the Homebridge RPi with **/bin/sh -l**:
`ssh -i /path/to/cert -o StrictHostKeyChecking=no admin@motioneye '/bin/sh -l /data/etc/enablecamera.sh'`

1gotroot, 2019-05-26:
Prior to upgrading to  20190427   I used to pull files off my RPi  with an  scp  command.   
Now I get      
sh: scp: No such file or directory

It certainly seems like the PATH is not set for sshd but a cat of /proc/sshd-pid/environ  shows an okay PATH.   Perhaps the daemon is changing its path to "".  Either way, the scp command that used to work no longer does.  

When this bug is fixed please indicate what the source of the error was - I'm curious to know what happened.  Thanks!

AIexBV, 2019-05-27:
I can confirm this issue. Remote execution of commands works only, if the command path is provided. Unfortunately scp has no option to provide a remote path. So my daily cron task is out of order currently. I hope this will be fixed soon.

ccrisan, 2019-05-27:
Guys, I'll look into this and see what went wrong. In the meantime, any further debugging information (such as step-by-step instructions on how to reproduce the problem) would be helpful.

AIexBV, 2019-05-28:
Debug info:
- First copy the public key of a remote machine to /data/etc/ssh_authorized_keys
- From the remote machine login to motioneyeos: "_ssh admin@meos_". That works, a password is not required.
- From the remote machine try to execute a command on the MotionEye machine e.g. "_ssh admin@meos ls_". That leads to an error:  "_sh: ls: No such file or directory_". 
- If you provide the path, the command works: "_ssh admin@meos **/bin/**ls_"

ccrisan, 2019-06-16:
Guys this should be fixed in the latest nightly builds.

closed 2019-06-16 by ccrisan

lmarceg, 2019-06-21:
I haven't understood what I should do to fix the issue. Where can I find the patch?

ccrisan, 2019-06-22:
Try the latest pre-release or the nightly build. There's nothing that you need to patch yourself.

kiwi-cam, 2019-06-23:
IMO, the update instructions aren't clear which might be what @lmarceg is struggling with.  For me the Web UI doesn't allow updates so you have to do them via SSH. Correct me if I'm wrong but this is the process you'll need to follow (sorry, don't have the resources to test the fix myself):

1. login via SSH (i.e. `ssh admin@motioneye`)
2. Run `fwupdate upgrade 20190619`. 20190619 is the current Pre-Release version found here: https://github.com/ccrisan/motioneyeos/releases.

There are other commands with fwupdate too:

- `fwupdate upgrade current` - installs the current release 
- `fwupdate versions` - lists the available release versions 
- or just run `fwupdate` to list all the options

ccrisan, 2019-06-23:
@kiwi-cam's answer is correct. Also I should have specified that if you want to upgrade to a prerelease (as opposed to a nightly build), you can do that (at your own risk) by enabling Prereleases in Expert Settings and then hitting a regular update via the Update button.

lmarceg, 2019-06-24:
Thank you very much for the info! I didn't know about the fwupdate command!
What I read is that I can update the current version to the nightly build or to the next pre-release, which may be less stable.
From the link, I see the pre-release is 20190619 while the nightly build is (I guess) dev20190620.
the `fwupdate versions` only shows 20190427 and I don't want to mess with a pre-release.
Maybe, as I have found in some other posts, the right command is `fwupdate upgrade https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20190620.img.xz` considering the fact I am using a Raspberry Pi 3B ?
Thanks for helping, it's the first time I face with this procedure and I want to make sure I do not disrupt anything.

lmarceg, 2019-07-03:
Since I didn't get any comments, I understood I had to work in a "roll-your-sleeves-up" way and I decided to try the nightly dev, which apparently is the best choice.
The one I saw is called dev20190620 and I initially tried `firmware upgrade dev20190620`. This didn't work, as this upgrade cannot be found, so I went the hard way with `fwupdate upgrade https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20190620.img.xz`.
It downloaded the image, installed it, rebooted but afther that, I couldn't access the Web GUI and no SSH was available. I guess my Raspberry was just stuck.
I have reflashed the backup I had previously prepared and now everything is back up again.
Now, I guess the only way is to wait for the latest stable release and do the official upgrade, maybe around december?

1gotroot, 2019-07-08:
Since I spent quite a bit of time trying to figure out what was the root cause of the lost functionality (but came up empty), I’d really appreciate hearing about what you did to fix this problem.
Thanks!!

kiwi-cam, 2019-07-08:
I've just updated my Raspberry Pi3 to motionEyeOS 20190619. I've removed the `/bin/sh -l ` that I had previously added as a workaround and my scripts are still working. Thanks @ccrisan, it certainly appears to be fixed.

I've also tested and confirmed that scp works correctly.

Interestingly, I updated using the same process @lmarceg discribed, `fwupdate upgrade https://github.com/ccrisan/motioneyeos/releases/download/20190619/motioneyeos-raspberrypi3-20190619.img.xz`, without any issues.

lmarceg, 2019-07-08:
@kiwi-cam , this is really interesting.
I will try to update my MotionEyeOS again, but this time I will try to read the logs from my SD Card in case it gets stuck and I will post the results here, so maybe I can also fix it.
I also spent some hours trying to fix the PATH problem like @1gotroot did, and I also couldn't find a solution. I need SSH to poweroff my MotionEye in case the UPS is just about to die and also to download/upload files with SCP.

lmarceg, 2019-08-26:
I would like to add the following comment, since I have just updated my Raspberry to the latest available nightly-patch, in order to solve the Chrome preview display bug.
After running the command, exactly like the first time, nothing happened: no webpage, and no ssh. I unplugged and replugged my Raspi, and everything loaded perfectly. This means that sometimes rebooting does not work and you need to do this process manually.
Just in case the same is happening to someone else, before re-flashing everything, just try to manually reboot the device!