motioneye-project/motioneyeos #1915: Add streamEye RTSP bitrate setting UI support

tigernero79, 2019-05-12:
I just updated motioneye to test the new streaming function via rtsp.

but I can't find a voice in the guide that tells me which url to use.

I tested such a solution

rtsp: // Tiger: password@192.168.x.x: 554

but it does not go.

what format does rtsp have url?

jasaw, 2019-05-12:
@tigernero79 Are you talking about testing RTSP stream output in Fast Network Camera mode? i.e. https://github.com/ccrisan/motioneyeos/pull/1764 ?

tigernero79, 2019-05-13:
yes, I installed the 20190511 release which should implement rest use.  In the guide I can't find anything about how to call up url.  from your GitHub I think we should call "rstp: // ddns_ip / h264" right?

jasaw, 2019-05-13:
@tigernero79 Thank you for your continued effort in testing.
Yes, the RTSP stream URL is: `rtsp://<rpi-IP-address>/h264`.
Please note that my PR adds support for RTSP mode in the backend only. UI support for RTSP mode does not exist yet. To enable RTSP mode, you'll have to manually edit some files. Please refer to my PR for more details on how to do that.

We'll have to discuss with @ccrisan about adding UI support for RTSP mode.

tigernero79, 2019-05-13:
ok thanks

tigernero79, 2019-05-13:
it would be very convenient to use a sort of digest authentication on rstp stream, for privacy.

I tested flow and it works great but the fact that you don't require a username and password is serious in my opinion.

url could be

rtsp: // admin: password @ ip_raspberry / h264

ccrisan, 2019-05-13:
Guys, commits c8ca30e25efe62147f03002d5b8572b22de31885 and 3c56bc2d75754234fc35c93158eab2578a3b96ff add basic UI support for @jasaw's work on RTSP. They should be already part of the latest nightly build (20190511), if my calculations are correct :)

closed 2019-05-13 by ccrisan

tigernero79, 2019-05-13:
I noticed the possibility to choose rstp and activate basic authentication.

wanting to reproduce it on kodi reference url with basic authentication is so?

rtsp: // user: password @ ip_raspberry / 264

because in the streameye file I notice that you have also entered the motioneyos value to the credentials

CREDENTIALS = "user: password: motionEyeOS"

jasaw, 2019-05-13:
I've created issue #1917 on adding basic auth to RTSP server.

Regarding the RTSP default port 554, it is hard-coded in the streameye script at the moment, but we can have the script pull this port number out of a config file. @ccrisan Would you be OK with configurable RTSP port number?

ccrisan, 2019-05-14:
@jasaw I don't see why not. We already have the "Streaming Port" option available for the FNC so we should use it.

jasaw, 2019-05-14:
@ccrisan We can reuse the existing streaming port key, but need to keep in mind that RTSP port should default to 554. Maybe creating a new key is better? Like RTSP port.

jasaw, 2019-05-15:
I've create issue #1918 for  supporting configurable RTSP port.