motioneye-project/motioneyeos #1920: email notifications issue - I keep getting "too many bad auth attempts" when I test the email

pmrid, 2019-05-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

pmrid, 2019-05-16:
Because Gmail doesn't seem to like MotionEyeOS, I switched to a couple of other email accounts and finally have been trying my Yahoo a/c.  But when I try to do a test email I get a "Too many bad auth attempts".  Can I reset this somehow?  Not sure where to go from here.

nickdkelly, 2019-06-06:
Gmail works just fine for me, so it may be worth checking whether you have 2-Factor authentication turned on for your Gmail account, and if so; generate an App password from within the 'Account Security' section of Gmail's settings. 

It's possible this is also the case for your Yahoo account? Though I'm unsure on that as I don't use Yahoo.

ilovelinux, 2019-06-18:
https://support.google.com/mail/answer/7126229