motioneye-project/motioneyeos #1921: Odroid XU4 Support for H264/OMX?

RatGit, 2019-05-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

RatGit, 2019-05-18:
**motionEyeOS Version:** 20190427
**Board Model:** Odroid XU4
**Camera:** 3 x 720P RTSP Network Cameras
**Network Connection:** Wired Ethernet

Dear MotionEyeOS People,

I've recently tried an Odroid XU4 SBC for this excellent product after previously successfully using R-Pi 3B+ boards. The reason for this was to support three 720P RTSP Cameras that the Pi was struggling with.

Everything has worked well on the XU4 except that H264/OMX support is not available for the FFMPEG binary provided with the XU4 OS image. AFAIK the Odroid XU4 GPU should be able to handle the HW encoding. Would it be possible to provide an FFMPEG binary for the XU4 with H264/OMX enabled?

jasaw, 2019-05-20:
I had a quick 10 minutes look into H264 encoding on the XU4. I don't own an Odroid XU4, so I'm not sure what is officially support, and what modifications are needed. I have a few questions though:
- From early discussions on various forums, it seem to suggest that it requires some feature to be enabled in the kernel. Is this still the case or is this enabled by default in the mainstream kernel?
- It also requires `h264_v4l2m2m` encoder to be enabled in ffmpeg. It may need Memeka's patches as well. Does anyone know if that's still the case? Does ffmpeg officially support `h264_v4l2m2m` encoder now?
If both kernel and ffmpeg don't require special hacks, it should be pretty easy to support it, provided the encoder is stable and behaves sensibly.

ccrisan, 2019-05-21:
@jasaw I can grant you remote login on my testing Odroid XU4 machine, running whatever OS you may be needing. Just let me know and I'll arrange it.

jasaw, 2019-05-22:
@ccrisan Thanks for the offer. I'll keep that in mind. I won't be doing any development on motionEyeOS for a couple of months because I'm way too busy with my work.

jasaw, 2019-07-01:
@ccrisan I have pulled in the necessary patches from memeka to enable h264 hw encoding on XU4 hardware. In your motion `camera-1.conf`, you'll need to select `h264_v4l2m2m` as your video encoder, similar to the omx version. You set bit rate the same way as omx version.
Since I don't have an XU4 board, I'm not able to test it. Can you please give it a whirl and let me know how you go?
My XU4 H264 branch is here:
https://github.com/jasaw/motioneyeos/tree/xu4-hw-h264

ccrisan, 2019-07-01:
@jasaw that's great news! Will definitely try it some time during the next weekend.

ccrisan, 2019-07-07:
@jasaw I can't seem to make Motion recognize `movie_codec mp4:h264_v4l2m2m`. Motion log says:

    [1:ml1:Camera1] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Invalid argument
    [1:ml1:Camera1] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
    [1:ml1:Camera1] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.

FFmpeg:

    # ffmpeg -codecs

    ffmpeg version 3.4.5 Copyright (c) 2000-2018 the FFmpeg developers
      built with gcc 8.2.1 (GNU Toolchain for the A-profile Architecture 8.2-2018.11 (arm-rel-8.26)
      configuration: --enable-cross-compile --cross-prefix=/home/ccrisan/motioneyeos-xu4/output/odroidxu4/host/bin/arm-linux-gnueabihf- --sysroot=/home/ccrisan/motioneyeos-xu4/output/odroidxu4/host/arm-buildroot-linux-gnueabihf/sysroot --host-cc='/home/ccrisan/motioneyeos-xu4/output/odroidxu4/host/bin/ccache /usr/bin/gcc' --arch=arm --target-os=linux --disable-stripping --pkg-config=/home/ccrisan/motioneyeos-xu4/output/odroidxu4/host/bin/pkg-config --disable-static --enable-shared --prefix=/usr --enable-avfilter --disable-version3 --enable-logging --enable-optimizations --disable-extra-warnings --enable-avdevice --enable-avcodec --enable-avformat --enable-network --disable-gray --enable-swscale-alpha --disable-small --enable-dct --enable-fft --enable-mdct --enable-rdft --disable-crystalhd --disable-dxva2 --enable-runtime-cpudetect --disable-hardcoded-tables --disable-mipsdsp --disable-mipsdspr2 --disable-msa --enable-hwaccels --disable-cuda --disable-cuvid --disable-nvenc --disable-avisynth --disable-frei0r --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-libdc1394 --disable-libgsm --disable-libilbc --disable-libvo-amrwbenc --disable-symver --disable-doc --enable-gpl --enable-nonfree --enable-ffmpeg --disable-ffplay --disable-ffserver --disable-avresample --disable-ffprobe --disable-postproc --enable-swscale --enable-indevs --enable-alsa --enable-outdevs --enable-pthreads --enable-zlib --disable-bzlib --disable-libfdk-aac --disable-libcdio --disable-gnutls --enable-openssl --disable-libdrm --disable-libopenh264 --enable-libvorbis --enable-muxer=ogg --enable-encoder=libvorbis --disable-vaapi --disable-vdpau --disable-mmal --disable-omx --disable-omx-rpi --disable-libopencv --disable-libopus --disable-libvpx --disable-libass --disable-libbluray --disable-librtmp --disable-libmp3lame --disable-libmodplug --disable-libspeex --enable-libtheora --disable-libwavpack --disable-iconv --disable-libfreetype --disable-fontconfig --disable-libopenjpeg --enable-libx264 --enable-libx265 --disable-x86asm --disable-mmx --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-avx2 --enable-armv6 --enable-vfp --enable-neon --disable-altivec --extra-libs=-latomic --enable-pic --cpu=cortex-a7
      libavutil      55. 78.100 / 55. 78.100
      libavcodec     57.107.100 / 57.107.100
      libavformat    57. 83.100 / 57. 83.100
      libavdevice    57. 10.100 / 57. 10.100
      libavfilter     6.107.100 /  6.107.100
      libswscale      4.  8.100 /  4.  8.100
      libswresample   2.  9.100 /  2.  9.100
    Codecs:
     D..... = Decoding supported
     .E.... = Encoding supported
     ..V... = Video codec
     ..A... = Audio codec
     ..S... = Subtitle codec
     ...I.. = Intra frame-only codec
     ....L. = Lossy compression
     .....S = Lossless compression
     -------
     D.VI.S 012v                 Uncompressed 4:2:2 10-bit
     D.V.L. 4xm                  4X Movie
     D.VI.S 8bps                 QuickTime 8BPS video
     .EVIL. a64_multi            Multicolor charset for Commodore 64 (encoders: a64multi )
     .EVIL. a64_multi5           Multicolor charset for Commodore 64, extended with 5th color (colram) (encoders: a64multi5 )
     D.V..S aasc                 Autodesk RLE
     D.VIL. aic                  Apple Intermediate Codec
     DEVI.S alias_pix            Alias/Wavefront PIX image
     DEVIL. amv                  AMV Video
     D.V.L. anm                  Deluxe Paint Animation
     D.V.L. ansi                 ASCII/ANSI art
     DEV..S apng                 APNG (Animated Portable Network Graphics) image
     DEVIL. asv1                 ASUS V1
     DEVIL. asv2                 ASUS V2
     D.VIL. aura                 Auravision AURA
     D.VIL. aura2                Auravision Aura 2
     ..V.L. av1                  Alliance for Open Media AV1
     D.V... avrn                 Avid AVI Codec
     DEVI.S avrp                 Avid 1:1 10-bit RGB Packer
     D.V.L. avs                  AVS (Audio Video Standard) video
     DEVI.S avui                 Avid Meridien Uncompressed
     DEVI.S ayuv                 Uncompressed packed MS 4:4:4:4
     D.V.L. bethsoftvid          Bethesda VID video
     D.V.L. bfi                  Brute Force & Ignorance
     D.V.L. binkvideo            Bink video
     D.VI.. bintext              Binary text
     D.VI.S bitpacked            Bitpacked
     DEVI.S bmp                  BMP (Windows and OS/2 bitmap)
     D.V..S bmv_video            Discworld II BMV video
     D.VI.S brender_pix          BRender PIX image
     D.V.L. c93                  Interplay C93
     D.V.L. cavs                 Chinese AVS (Audio Video Standard) (AVS1-P2, JiZhun profile)
     D.V.L. cdgraphics           CD Graphics video
     D.VIL. cdxl                 Commodore CDXL video
     D.V.L. cfhd                 Cineform HD
     DEV.L. cinepak              Cinepak
[... cut ...]
     D.S... stl                  Spruce subtitle format
     DES... subrip               SubRip subtitle (decoders: srt subrip ) (encoders: srt subrip )
     D.S... subviewer            SubViewer subtitle
     D.S... subviewer1           SubViewer v1 subtitle
     DES... text                 raw UTF-8 text
     D.S... vplayer              VPlayer subtitle
     DES... webvtt               WebVTT subtitle
     DES... xsub                 XSUB

Am I doing something wrong? Do you need more logs/info? Would you like remote access on my test XU4?

jasaw, 2019-07-08:
> DEV.LS h264                 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (decoders: h264 h264_v4l2m2m ) (encoders: libx264 libx264rgb h264_v4l2m2m )

`h264_v4l2m2m` encoder is definitely compiled in.

Looks like motion found the encoder, went ahead to open the encoder (`avcodec_open2` function call), but the encoder doesn't like the arguments passed into it. I suspect it doesn't like something in the `ctx_codec` struct. Question is what doesn't it like.
 Can you please bump up the log level and verify that you have `Using codec h264_v4l2m2m` debug output from motion? It should also print out error messages from ffmpeg.

I'll also play with h264_v4l2m2m on my RPi3, see if it behaves the same way.

jasaw, 2019-07-08:
@ccrisan I'm getting segfault on my RPi3 when I use h264_v4l2m2m encoder. Let's see what's tripping your  h264_v4l2m2m on XU4.

Can you please post the error messages before and after the `[1:ml1:Camera1] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Invalid argument` line? Anything after the `[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec h264_v4l2m2m` line is of interest. If you run motion in the foreground with log level 8, it should give us error messages, something like `motion -n -c camera-1.conf -d 8`. Enable `emulate_motion on` in your camera-1.conf file to trigger recording straight away.

I might need remote access to your XU4 at a later time if that's not too difficult to set up.

ccrisan, 2019-07-14:
@jasaw I just dropped you an email with remote login details on my XU4. Keep us posted.

jasaw, 2019-07-15:
@ccrisan Thank you for allowing remote access to your XU4. I'm taking a look at the problem now.

jasaw, 2019-07-15:
@ccrisan Here's a quick update on my findings.
ffmpeg failed to initialize the video encoder because we are missing the Samsung firmware file. dmesg output below:

>[38618.087183] s5p-mfc 11000000.codec:: Direct firmware load for s5p-mfc-v8.fw failed with error -2
[38618.087192] s5p_mfc_load_firmware:70: Firmware is not present in the /lib/firmware directory nor compiled in kernel

I downloaded `s5p-mfc-v8.fw` from https://github.com/armbian/firmware/raw/master/s5p-mfc-v8.fw and placed it in `/lib/firmware` directory. Now ffmpeg triggers a different error in dmesg:

>[40022.156171] s5p_mfc_wait_for_done_dev:34: Interrupt (dev->int_type:0, command:12) timed out
[40022.156179] s5p_mfc_init_hw:244: Failed to load firmware
[40022.660730] s5p_mfc_bus_reset:110: Timeout while resetting MFC.
[40024.676195] s5p_mfc_wait_for_done_dev:34: Interrupt (dev->int_type:0, command:12) timed out
[40024.676203] s5p_mfc_init_hw:244: Failed to load firmware
[40025.180709] s5p_mfc_bus_reset:110: Timeout while resetting MFC.

I checked [Memeka's 4.9.x kernel](https://github.com/mihailescu2m/linux/commits/odroidxu3-4.9.y) patches and suspect we need all the patches as well. Now we have 2 choices:
1. Upgrade to kernel 4.14. Someone reported that MFC video encoder works on 4.14 : https://forum.armbian.com/topic/8451-howto-emby-server-with-hardware-transcoding-in-xu4hc1hc2-armbian-stretch/
2. We pull all the patches from Memeka's repository into our own.

What's your preference? How hard is it for you to upgrade it to kernel 4.14 ?

ccrisan, 2019-07-15:
I think the best way to go is to upgrade to 4.14. Hardkernel seem to maintain this branch for the XU4: https://github.com/hardkernel/linux/tree/odroidxu4-4.14.y

jasaw, 2019-07-16:
@ccrisan I compiled a new XU4 image with 4.14 kernel from https://github.com/hardkernel/linux/archive/4.14.127-164.tar.gz and loaded it onto your test XU4, but I can't access it anymore. Oops, sorry. Can you please take a look at why 4.14 kernel is not working? I can't do it remotely. :-(

ccrisan, 2019-07-16:
@jasaw apparently the XU4 with 4.14 does not properly honor the `reboot` command. The watchdog doesn't seem to do its job either.

This being said, two manual power cycles later and the XU4 is updated :) You should now be able to access it again, just like before. If you need any kind of rebooting, just let me know and I'll power-cycle it manually as soon as I get home.

We'll need to sort out this rebooting issue somehow but that's for another day.

jasaw, 2019-07-17:
@ccrisan I'm still getting fundamental issue of loading the samsung firmware with 4.14 kernel. I'm not sure what's wrong with it yet. I get this in dmesg when ffmpeg attempts to use the hardware encoder.
>[18991.106094] s5p_mfc_wait_for_done_dev:34: Interrupt (dev->int_type:0, command:12) timed out
[18991.106102] s5p_mfc_init_hw:249: Failed to load firmware
[18991.610059] s5p_mfc_bus_reset:115: Timeout while resetting MFC.
[18993.634083] s5p_mfc_wait_for_done_dev:34: Interrupt (dev->int_type:0, command:12) timed out
[18993.634088] s5p_mfc_init_hw:249: Failed to load firmware
[18994.138056] s5p_mfc_bus_reset:115: Timeout while resetting MFC.

This is the command I used to test:
```
cd /home/ftp/sdcard/Camera1/2019-07-15
rm test.mp4; ffmpeg -i 05-42-54.mp4 -codec:v copy -bsf:v h264_mp4toannexb -f rawvideo - | ffmpeg -loglevel debug -r 24 -i - -an  -vcodec h264_v4l2m2m -b:v 2M -profile:v 10 -pix_fmt nv21 test.mp4
```
and got this error from ffmpeg because the driver couldn't load the firmware.
>[h264_v4l2m2m @ 0xdd5b0] Could not find a valid device

Again, I've downloaded `s5p-mfc-v8.fw` from [armbian](https://github.com/armbian/firmware/raw/master/s5p-mfc-v8.fw) and placed it in `/lib/firmware` directory. I'm not sure which is the official channel to get the samsung firmware. On debian, it's part of the `firmware-samsung` package.

I've also checked the `/boot/exynos5422-odroidxu4.dtb` file, which is compiled from `linux-custom/arch/arm/boot/dts/exynos5420.dtsi` and `linux-custom/arch/arm/boot/dts/exynos5422-odroidxu4.dts`. Mali GPU info is definitely in the dtsi file.

Google search didn't yield a lot of results. Any idea on what else to try? What's the difference between our OS and Armbian? Presumably the firmware loads correctly on Armbian.

jasaw, 2019-07-25:
@ccrisan Did you get a chance to look into the firmware loading issue? I seem to have lost remote access to your XU4.

I checked the Ambian s5p_mfc_v8 firmware against the one in linux-firmware-20190618, and they are the same. Have you tried Ambian OS?

ccrisan, 2019-07-26:
@jasaw not yet. I'll try to see what's going on during this weekend.

ccrisan, 2019-07-26:
I found a first issue, but we're still far away from a solution. The firmware file was an HTML that needed redirection:

    [root@meye-06323001 firmware]# cat s5p-mfc-v8.fw 
    <html><body>You are being <a href="https://raw.githubusercontent.com/armbian/firmware/master/s5p-mfc-v8.fw">redirected</a>.</body></html>[root@meye-06323001 firmware]

I downloaded the actual file and now `dmesg` tells me:

    [    1.731997] s5p-mfc 11000000.codec: preallocated 16 MiB buffer for the firmware and context buffers
    [    1.732112] (NULL device *): Direct firmware load for s5p-mfc-v8.fw failed with error -2
    [    1.732124] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory nor compiled in kernel
    [    1.732308] s5p-mfc 11000000.codec: decoder registered as /dev/video10
    [    1.732455] s5p-mfc 11000000.codec: encoder registered as /dev/video11

Will get back with more findings...

jasaw, 2019-07-27:
Make sure the s5p-mfc-v8.fw file is placed in /lib/firmware directory.
From your dmesg, you have encoder and decoder registered as /dev/video10 and 11, but question is which encoder is that.

Try encoding something from ffmpeg command, something like this:
```
cd /home/ftp/sdcard/Camera1/2019-07-15
rm test.mp4; ffmpeg -i 05-42-54.mp4 -codec:v copy -bsf:v h264_mp4toannexb -f rawvideo - | ffmpeg -loglevel debug -r 24 -i - -an  -vcodec h264_v4l2m2m -b:v 2M -profile:v 10 -pix_fmt nv21 test.mp4
```

This will trigger the driver to load the s5p-mfc-v8 firmware again and you can check dmesg to see what's wrong this time.

ccrisan, 2019-07-27:
@jasaw the XU4 is back online, running Armbian. I emailed you the details.

jasaw, 2019-07-27:
@ccrisan I seem to have lost connection to your XU4 again. From the brief moment that I managed to get on, I've checked a few things.
The dmesg error that you get is normal:
>[    1.732112] (NULL device *): Direct firmware load for s5p-mfc-v8.fw failed with error -2
[    1.732124] s5p_mfc_load_firmware:73: Firmware is not present in the /lib/firmware directory

The driver tried to load the firmware but the filesystem probably has been mounted yet.
When ffmpeg attempts to use the encoder later, the driver will load the firmware again.

I installed ffmpeg from apt, and found that they're running a much newer version, but not sure if that's necessary yet. I also noticed their ffmpeg version supports hevc_v4l2m2m. I'm interested in seeing what kind of performance we get with hardware hevc encoder.

ccrisan, 2019-07-27:
@jasaw the XU4 goes down from time to time following the trends of the electricity around here :)

Good to hear those lines are normal. That's what we get with motionEyeOS if the proper firmware is where it should be. If you think we should upgrade ffmpeg, just let me know.

jasaw, 2019-07-27:
With armbian kernel 4.14.127-odroidxu4, it managed to load the s5p-mfc-v8 firmware:
>[h264_v4l2m2m @ 0x48b120] probing device /dev/video11
[h264_v4l2m2m @ 0x48b120] driver 's5p-mfc' on card 's5p-mfc-enc'
[h264_v4l2m2m @ 0x48b120] Using device /dev/video11
[h264_v4l2m2m @ 0x48b120] driver 's5p-mfc' on card 's5p-mfc-enc'
[h264_v4l2m2m @ 0x48b120] output : NM21 16 buffers initialized: 0960x0720, sizeimage 00691456, bytesperline 00000960
[h264_v4l2m2m @ 0x48b120] capture: H264 04 buffers initialized: 0000x0000, sizeimage 00532480, bytesperline 00532480

At the same time, I get this error on dmesg:
>[ 1110.499680] vidioc_try_fmt:1070: failed to try output format
[ 1110.499690] vidioc_try_fmt:1070: failed to try output format

I don't know what that error means yet.

I forced motion to use h264_v4l2m2m encoder and got a segfault when it starts recording. I haven't got time to track it down yet.

Just for the record, armbian is running ffmpeg version 4.1.3-1. At this stage, I don't think it matters which ffmpeg version we run.

Anyway, I'm gonna get some sleep now, continue tomorrow. If you want to take a look, I've put motion source and test config in /root directory.

jasaw, 2019-07-28:
I'm getting segfault in the ffmpeg library from armbian.

From the logs, motion used h264_v4l2m2m encoder, ffmpeg initialized it successfully, configured the encoder parameters (everything looks good here). As soon as motion sends a frame to it, it segfaults. I don't know why it's segfaulting. I'll need to compile our own ffmpeg library with more debug to find out where it's segfaulting.

>[1:ml1] [INF] [EVT] event_new_video: Source FPS 20
[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec h264_v4l2m2m
[1:ml1] [INF] [ENC] ffmpeg_set_quality: h264_v4l2m2m codec vbr/crf/bit_rate: 2700000
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: probing device /dev/video0
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: driver 'uvcvideo' on card 'UVC Camera (046d:081b)'
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: v4l2 output format not supported
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: probing device /dev/video21
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: driver 'exynos-gsc' on card 'exynos-gsc gscaler'
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: v4l2 capture format not supported
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: probing device /dev/video20
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: driver 'exynos-gsc' on card 'exynos-gsc gscaler'
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: v4l2 capture format not supported
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: probing device /dev/video11
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: driver 's5p-mfc' on card 's5p-mfc-enc'
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Using device /dev/video11
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: driver 's5p-mfc' on card 's5p-mfc-enc'
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: output : NM21 16 buffers initialized: 0960x0720, sizeimage 00691456, bytesperline 00000960
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: capture: H264 04 buffers initialized: 0000x0000, sizeimage 00532480, bytesperline 00532480
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: number of B-frames = 0
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: header mode = 0
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: bit rate = 2700000
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: gop size = 10
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder Context: id (27), profile (100), frame rate(0/1), number b-frames (0), gop size (10), bit rate (2700000), qmin (2), qmax (31)
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: h264 profile = 4
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder adjusted: qmin (0), qmax (51)
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: minimum video quantizer scale = 0
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Encoder: maximum video quantizer scale = 51
[1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Setting default whitelist 'file,crypto'
[1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /root/motioneye/2019-07-28/14-27-48.mp4
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1:ml1] [INF] [ENC] ffmpeg_encode_video: avcodec_send_frame encode
Segmentation fault

jasaw, 2019-07-29:
@ccrisan I've tracked the segfault down to dereferencing of a pointer that's passed from motion software. The pointer looks sensible, but don't know why it segfaults. This is going to need more investigation.
I've reported the issue to motion project, hoping to get some guidance: https://github.com/Motion-Project/motion/issues/977

jasaw, 2019-08-13:
@ccrisan I found out that motion is not setting up ffmpeg buffers correctly by calling `av_frame_get_buffer`, which caused the earlier segfault when dereferencing `buf[0]->size`. I fixed that in motion and now getting segfault when dereferencing `buf[1]->size`. This segfault is not motion's bug anymore. I get this same segfault even with ffmpeg's own example code. I've also tried various versions of ffmpeg, with and without Memeka's patch. Anyway, I've reported the problem here: https://trac.ffmpeg.org/ticket/8079#ticket  Let's see what comes out of it.

ffmpeg example code from the master branch:
>./encode_video /root/motioneye/test.mp4 h264_v4l2m2m
[h264_v4l2m2m @ 0x14993b0] driver 'uvcvideo' on card 'UVC Camera (046d:081b)'
[h264_v4l2m2m @ 0x14993b0] driver 'exynos-gsc' on card 'exynos-gsc gscaler'
[h264_v4l2m2m @ 0x14993b0] driver 'exynos-gsc' on card 'exynos-gsc gscaler'
[h264_v4l2m2m @ 0x14993b0] driver 's5p-mfc' on card 's5p-mfc-enc'
[h264_v4l2m2m @ 0x14993b0] Using device /dev/video11
[h264_v4l2m2m @ 0x14993b0] driver 's5p-mfc' on card 's5p-mfc-enc'
[h264_v4l2m2m @ 0x14993b0] Encoder does not support b-frames yet
[h264_v4l2m2m @ 0x14993b0] h264 profile not found
[h264_v4l2m2m @ 0x14993b0] Encoder adjusted: qmin (0), qmax (51)
Send frame   0
ffmpeg: avcodec_send_frame called
ffmpeg: calling codec->send_frame
v4l2_m2m_enc: v4l2_send_frame called
v4l2_context: ff_v4l2_context_enqueue_frame called
calling ff_v4l2_buffer_avframe_to_buf
v4l2_buffers: ff_v4l2_buffer_avframe_to_buf called, out ptr: 0x149b4a0
num_planes = 2
0: v4l2_bufref_to_buf
0: frame ptr = 0x149ea70
0: &frame->buf[0] = 0x149eb68
0: frame->buf[0] = 0x149ec70
0: frame->buf[0]->size = 156800
0: frame->buf[0]->data ptr = 0xb654a020
1: v4l2_bufref_to_buf
1: frame ptr = 0x149ea70
1: &frame->buf[1] = 0x149eb6c
1: frame->buf[1] = (nil)
Segmentation fault

ffmpeg command line program from the master branch too:
>ffmpeg -loglevel debug -i /dev/video0 -c:v h264_v4l2m2m -b:v 2M -profile:v 66 /root/motioneye/test.mp4
ffmpeg version N-94529-g3a557c5d88 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration:
  libavutil      56. 33.100 / 56. 33.100
  libavcodec     58. 55.100 / 58. 55.100
  libavformat    58. 30.100 / 58. 30.100
  libavdevice    58.  9.100 / 58.  9.100
  libavfilter     7. 58.100 /  7. 58.100
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'h264_v4l2m2m'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '2M'.
Reading option '-profile:v' ... matched as option 'profile' (set profile) with argument '66'.
Reading option '/root/motioneye/test.mp4' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x1853370] fd:3 capabilities:84200001
[video4linux2,v4l2 @ 0x1853370] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x1853370] Querying the device for the current frame size
[video4linux2,v4l2 @ 0x1853370] Setting frame size to 960x720
[video4linux2,v4l2 @ 0x1853370] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1853370] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1853370] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1853370] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1853370] The V4L2 driver changed the pixel format from 0x32315659 to 0x56595559
[video4linux2,v4l2 @ 0x1853370] Trying to set codec:rawvideo pix_fmt:yuv422p
[video4linux2,v4l2 @ 0x1853370] The V4L2 driver changed the pixel format from 0x50323234 to 0x56595559
[... 53 lines cut ...]
Output #0, mp4, to '/root/motioneye/test.mp4':
  Metadata:
    encoder         : Lavf58.30.100
    Stream #0:0, 0, 1/10240: Video: h264 (h264_v4l2m2m) (Baseline), 1 reference frame (avc1 / 0x31637661), yuyv422, 960x720, 0/1, q=2-31, 2000 kb/s, 10 fps, 10240 tbn, 10 tbc
    Metadata:
      encoder         : Lavc58.55.100 h264_v4l2m2m
Clipping frame in rate conversion by 0.000008
ffmpeg: avcodec_send_frame called
ffmpeg: calling codec->send_frame
v4l2_m2m_enc: v4l2_send_frame called
v4l2_context: ff_v4l2_context_enqueue_frame called
calling ff_v4l2_buffer_avframe_to_buf
v4l2_buffers: ff_v4l2_buffer_avframe_to_buf called, out ptr: 0x185c860
num_planes = 2
0: v4l2_bufref_to_buf
0: frame ptr = 0x1860870
0: &frame->buf[0] = 0x1860968
0: frame->buf[0] = 0x1859560
0: frame->buf[0]->size = 1382400
0: frame->buf[0]->data ptr = 0xb6c78000
1: v4l2_bufref_to_buf
1: frame ptr = 0x1860870
1: &frame->buf[1] = 0x186096c
1: frame->buf[1] = (nil)
Segmentation fault

ccrisan, 2019-08-13:
@jasaw thanks for keeping us up to date with the progress. If you feel like we need to provide a patch to ffmpeg as well, we could do that :)

jasaw, 2019-08-15:
Just a quick update.

Turns out that the samsung v4l2m2m driver only accepts multi-planar image data, but `av_frame_get_buffer` from ffmpeg only allocates one buffer rather than 2, which causes the segfault.  I hacked the `av_frame_get_buffer` function to always allocate 2 buffers and that got rid of the segfault for now. Obviously this is not the correct way of fixing it, but let's stop it from crashing first. I tried recording video from motion, but only got a green screen. I don't know what's wrong yet, maybe video format or YUV data not packed correctly.

FFmpeg's encode_video example hasn't been updated for a while, and it really is not a very good example. I'm hacking it so that I can get something working in a small code base, but I'm getting segfault when writing encoded frame to output file via `av_write_frame` call.

That's where I'm up to.

jasaw, 2019-08-16:
@ccrisan I managed to record successfully using h264_v4l2m2m encoder, but I can't verify whether the CrCb colour channels are correct (it's almost pitch black at your place). Again, I hacked ffmpeg and motion to get this to work. Once we can verify the encoded video is correct, we'll have to implement the fix properly.

jasaw, 2019-08-16:
@ccrisan Currently I modified motion to always allocate multi-planar image buffers for ffmpeg to pass to the Samsung driver, which stops the segfault. The consequence of this is one additional memcpy to get the raw image across to this multi-planar image buffer. I'm also converting YUV420 format to NV21 format in motion by interleaving the U and V bytes. Now there's some sunlight at your place, I can verify that the h264_v4l2m2m is indeed working. If you want to play around with it, you can try running `/root/motion-release-4.2.2/motion -c /root/camera-temp.conf -d 9 -n`, which will record constantly to `/root/motioneye` directory. Be warned that it is VERY verbose at the moment because I've sprinkled debug print everywhere. It's just a proof of concept. :-)

We should probably look at how to clean this up. I'm not sure how to fix it properly yet.
- Motion should automatically detect the format supported by the encoder, then convert YUV420 to NV21 via an ffmpeg function.
- I'm not sure where to fix the multi-planar causing segfault issue. FFmpeg should probably be fixed to NOT segfault, but I'm not putting my hopes on ffmpeg developers to do anything about my bug report. That's why I'm working around it in motion by always allocating the buffers for multi-planar.

Any ideas are welcomed.

jasaw, 2019-08-16:
There seems to be another problem. The encoded output seems to be missing a keyframe. When I run ffprobe on the output file, I get 
>st: 0 edit list: 1 Missing key frame while searching for timestamp: 0

This causes some video players to be very unhappy. Plays fine on some.

I don't know how to fix it.

jasaw, 2019-08-17:
The missing key frame at timestamp 0 appears to be caused by the h264_v4l2m2m encoder outputting some header as the first frame. The header bytes look like this:
>[ 00 00 00 01 67 64 00 0a ac b4 0b 04 b2 00 00 00 01 68 ce 01 9e 0c ]

I don't know what they mean yet, but I suspect ffmpeg setting HEADER_MODE_SEPARATE in the encoder may have something to do with it.

jasaw, 2019-08-19:
`HEADER_MODE_SEPARATE` is indeed the cause of the missing key frame problem. In this separate header mode, the encoder first spits out the stream header (which contains [NAL units](https://yumichan.net/video-processing/video-compression/introduction-to-h264-nal-unit/)) as the first packet with pts 0, then spits out the first encoded frame (which is also a key frame).

I see 2 solutions to this:
1. Change `HEADER_MODE_SEPARATE` to `HEADER_MODE_JOINED_WITH_1ST_FRAME`. I tried this, but ffmpeg locks up waiting forever for data from the encoder. I added a 200ms timeout in ffmpeg and managed to output a properly formed media file.
2. Figure out a way to detect which packet is header, which is image, and properly write the header information to the output media file.

jasaw, 2019-08-19:
@ccrisan I finally got h264_v4l2m2m working on your XU4 running Armbian. I had to make slight changes to ffmpeg to get "joined with 1st frame" header mode to work. On motion side, I had to make more changes, basically handle h264_v4l2m2m a bit differently in a few places. Can you please have a play with it, let me know if it actually works (I only tested with encoding static frames since I'm developing remotely)? `/root/motion-release-4.2.2/motion -c /root/camera-temp.conf -d 9 -n` should get you going. Recorded videos are stored in `/root/motioneye` directory.
At some point, we should let this system run for 1 week to check stability. Let me know when you want to run this stability test and I won't mess with the system.

For your reference, all my work is here: https://github.com/jasaw/motioneyeos/tree/xu4-hw-h264

This leaves us with one last problem: Unable to load the s5p-mfc-v8 firmware on MotionEyeOS, but works fine on Armbian.

Just a few notes: All this stuff is highly experimental, and I haven't checked whether my changes to motion broke anything on other platforms. I also need to upstream the motion patches, which means our patches now are nowhere near final.

ccrisan, 2019-08-19:
@jasaw let's start the long test right away. I don't play with my XU4 these days so just let it do its stuff. There's one single problem: electricity may go off every now and then so you'll probably need to set things to start up automatically (probably using `/etc/rc.local` or something), just in case.

> This leaves us with one last problem: Unable to load the s5p-mfc-v8 firmware on MotionEyeOS, but works fine on Armbian.

I'll take care of that.

> Just a few notes: All this stuff is highly experimental, and I haven't checked whether my changes to motion broke anything on other platforms. I also need to upstream the motion patches, which means our patches now are nowhere near final.

Let's apply these patches only on the XU4 build, at least for now. This way, we reduce the possible negative impact to the 0.1% of the users (that use the XU4). I am one of them and I'll make sure to report back any issues.

jasaw, 2019-08-20:
@ccrisan I have installed motioneye on your XU4 and it should start up automatically. It is configured to save movies to `/var/lib/motioneye` on motion trigger and remove files after 1 day. If you are physically near the camera, can you please trigger some motion every once in a while to test?

ccrisan, 2019-08-21:
@jasaw will see what I can do about motion. Can you configure the threshold to a low value so that motion is often detected even though nothing really moves around?

jasaw, 2019-08-21:
@ccrisan I've set the motion threshold quite low, just not sure how low I should go. I don't want to fill up the storage until the device fails to boot. It has recorded a few videos so far, and looks pretty good to me.

I will send it emulate motion every now and then to trigger recording.

jasaw, 2019-08-28:
@ccrisan I've just discovered another issue with this encoder. The encoded video doesn't skip the "motion gap" sections. I suspect this encoder is ignoring the PTS in the input frame. I'll think about a workaround for it. Probably mess with the encoded PTS field.

jasaw, 2019-08-29:
Turns out the static images recorded during "motion gap" is caused by a bug in motion. It has nothing to do with this encoder. The encoder does work properly after I fixed this bug in motion. See https://github.com/Motion-Project/motion/pull/997

jasaw, 2019-09-02:
@ccrisan There's one small issue left with the h264_v4l2m2m encoder: There's one occasion where there was a big change in the frames that caused blocky video, but the encoder did not recover from the blockiness. Usually it recovers, but in this one instance, it did not. I think this is deep within samsung firmware which I don't be able to fix.

ccrisan, 2019-09-02:
@jasaw what does it actually do when that happens? Does it block and the whole video encoding process is dead until reboot?

jasaw, 2019-09-02:
It doesn't block the process. You just get a lower quality video, looks as though bit rate is too low.

ccrisan, 2019-09-02:
@jasaw do we have a workaround for this issue? Like detecting it somehow and rebooting the system?

jasaw, 2019-09-02:
@ccrisan There's no need to reboot the system. The encoder recovers the next time we encode a video.

ccrisan, 2019-09-02:
@jasaw that sounds reasonable enough. So the only real issue left here is updating motionEyeOS to include and load the correct firmware?

jasaw, 2019-09-03:
@ccrisan This video `/root/motioneye/19-41-49.mp4` on your XU4 that shows the bit rate problem in the samsung encoder. The video started out at perfect quality, but after the camera moved, the video became blocky. Motion sets the GOP to 1 second, and forcefully set the keyframe once per second, but I don't see keyframe in the encoded video. If there's a keyframe after camera move, the video would have recovered from the blockiness. I'll look into ffmpeg code to see if keyframe is ignored for v4l2 encoders.

Other than this, yes, only issue left is the include and load firmware in motionEyeOS.

jasaw, 2019-09-03:
I used ffprobe to print every frame in the encoded video. Turns out that every frame is a keyframe . Weird.

ccrisan, 2019-09-03:
@jasaw sure! My apologies, I needed the USB camera for other tests but I forgot to plug it in back. It's now back in place.

jasaw, 2019-09-04:
@ccrisan On your XU4, I have set v4l2 header mode back to separate in ffmpeg, added some code in motion to inject the NAL units from the encoder into the first I frame. The encoded video file seems to show I frames and P frames, which is good. I'm going to let it run for a while to see if we still get the blocky video issue.
If you happen to walk past your camera, can you please move the camera a bit to trigger high bit rate in encoded video? This will tell us whether we still have blocky video or not.

ccrisan, 2019-09-04:
@jasaw I'll try to place the camera facing the yard so it generates motion whenever Max (the dog) moves around :)

jasaw, 2019-09-10:
@ccrisan I do get a lot of videos generated now. I've been checking the videos every now and then, and the videos look pretty good. The XU4 encoder output quality is not as good as the RPi OMX, but I don't think there's anything we can do to the encoding algorithm. Let's roll this out for other users to test and get some feedback. What do you think?

ccrisan, 2019-09-11:
@jasaw yeap, I'll try to put together everything that is needed so that users can test this on the XU4 using our nightly builds.

My plans are however to release a stable version first with everything that has changed since April. I think we're very close to a new stable release but I still need some feetback on recent prereleases before deeming it stable.

jasaw, 2019-09-18:
@ccrisan Looks like the XU4 encoder is still getting itself into the low bitrate blocky video state ocassionally. I checked the encoded video frame by frame, and looks like it only outputs I frames (key frames), but no P frames. I don't know how to trigger it, and which component is responsible for this bug. I'll have to report this bug to ffmpeg developers and see if they have any insights. I think we still proceed with rolling this out to other users and mark this as a known bug.

In the mean time, if anyone has idea on how to track this down, please chime in. 

This is the badly encoded video.
> root@odroidxu4:~/motion-release-4.2.2# ffprobe -show_frames /var/lib/motioneye/Camera1/2019-09-17/19-21-56.mp4 | grep -E 'key_frame|pkt_pts|pict_type' | head -20
ffprobe version 4.2 Copyright (c) 2007-2019 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration:
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/var/lib/motioneye/Camera1/2019-09-17/19-21-56.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.29.100
  Duration: 00:00:41.91, start: 0.000000, bitrate: 1608 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p, 1280x960, 1606 kb/s, 13.96 fps, 16384 tbr, 16384 tbn, 32768 tbc (default)
    Metadata:
      handler_name    : VideoHandler
key_frame=1
pkt_pts=0
pkt_pts_time=0.000000
pict_type=I
key_frame=1
pkt_pts=1035
pkt_pts_time=0.063171
pict_type=I
key_frame=1
pkt_pts=2068
pkt_pts_time=0.126221
pict_type=I
key_frame=1
pkt_pts=3868
pkt_pts_time=0.236084
pict_type=I
key_frame=1
pkt_pts=4902
pkt_pts_time=0.299194
pict_type=I

This is a good video.
> root@odroidxu4:~/motion-release-4.2.2# ffprobe -show_frames /var/lib/motioneye/Camera1/2019-09-17/19-24-00.mp4 | grep -E 'key_frame|pkt_pts|pict_type' | head -20
ffprobe version 4.2 Copyright (c) 2007-2019 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration:
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/var/lib/motioneye/Camera1/2019-09-17/19-24-00.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.29.100
  Duration: 00:01:24.79, start: 0.000000, bitrate: 5458 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p, 1280x960, 5456 kb/s, 15.07 fps, 15360 tbr, 15360 tbn, 30720 tbc (default)
    Metadata:
      handler_name    : VideoHandler
key_frame=1
pkt_pts=0
pkt_pts_time=0.000000
pict_type=I
key_frame=0
pkt_pts=1217
pkt_pts_time=0.079232
pict_type=P
key_frame=0
pkt_pts=2026
pkt_pts_time=0.131901
pict_type=P
key_frame=0
pkt_pts=3016
pkt_pts_time=0.196354
pict_type=P
key_frame=0
pkt_pts=4054
pkt_pts_time=0.263932
pict_type=P

ccrisan, 2019-09-29:
@jasaw I have just marked 20190911 as stable and updated the master branch accordingly. We can now continue the development on `dev`. I'm going to merge `xu4-hw-h264` into `dev` as well.

Apart from loading the Samsung firmware, what else do we need to make sure to do about the XU4 and its HW acceleration? I believe we should also add a dedicated experimental codec option in motionEye, don't we?

jasaw, 2019-09-29:
@ccrisan You're right, we only need to resolve the loading Samsung firmware issue and add h264_v4l2m2m to motionEye.

All my work on motion to support v4l2m2m has been merged upstream. I was going to upgrade motion in `xu4-hw-h264` branch to the latest master until I encountered compile problem. At some point, we should upgrade to the latest motion master and update the patch files (I have already updated the patch file on my branch https://github.com/jasaw/motioneyeos/commit/3bebed32fb6bc27fac501354587a539d50ab0ee2). Let me know how you want to proceed with upgrading motion.

After upgrading to the latest motion, can you please also test **without** these 2 patches to see whether the encoder ever locks up. I prefer to maintain less patches going forward.
- package/ffmpeg/1006-avcodec-v4l2-add-timeout-when-dequeueing-buffers.patch
- package/motion/1001-decouple-avcodec-send-and-receive.patch

ccrisan, 2019-09-30:
Guys, compiling latest dev fails with

    gstav.c:33:10: fatal error: libavfilter/avfiltergraph.h: No such file or directory

Any idea what could be the problem?

jasaw, 2019-10-01:
@ccrisan Weird. I have just pulled from latest dev branch, did a clean build with no error.
Anyway, your error message looks like it's failing at gstreamer package. Wouldn't hurt to upgrade gstreamer to a newer version to see if it fixes it for you (refer to my commit: https://github.com/jasaw/motioneyeos/commit/d69dab0acbdc6658879b183dfedbda30ae0acb61). Only problem is the newer gstreamer seems to have latency issue with RTSP in FNC mode as reported by @popoviciri. Hopefully it's not too long before we can replace gstreamer with my v4l2multi_stream_mmal.

jasaw, 2019-10-01:
@ccrisan My bad, the compilation indeed failed for raspberrypi builds. I've just created a PR to upgrade gstreamer to version 1.16.0. Please review: upgrade-gstreamer

[22 more replies]

closed 2020-01-14 by ccrisan