motioneye-project/motioneyeos #1922: Pi locks up when turning Movies Off?

alphanumeric007, 2019-05-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

alphanumeric007, 2019-05-18:
I have read the contributing guide.
I have read the FAQ
I am running Motion Eye OS 20190119
Motion version 4.1.1
Raspberry Pi 3A+
Pi foundation camera V2
Connected via WIFI
Running headless with no USB devices connected.

With my video resolution set at 1280x720 and frame rate set at 30 usually I have no problem recording videos. Movies on, continuous recording, H.264(.mp4), 75% movie quality. I do get the occasional freeze lockup. Usually when I try to stop the recording.
If I switch to 1920 x 1080 its almost guarantied that Pi will lock up and or become unresponsive via the web interface. I can't stop the recording, reboot or shut down. If I power cycle the Pi its usually usable again but the recorded video is corrupted. I get an unsupported error when I try to play it back. Some times the lockup happens as soon as I select apply? 
I usually end up reimaging my SD card to get things fully functional again. What am I doing wrong? 
Also, I have yet to ever get Motion Eye to work on my 5GHZ WIFI?  
16 GB Lexar 300x Micro SD card. Also happens with 16 GB Sandisk class 10 Micro SD cards.

RatGit, 2019-05-18:
I suspect that you may be pushing the Pi a bit hard with 1920x1080... 
Note: I found that my Pi 3B+ couldn't handle three 720P cameras, so I switched to an Odroid U4 which can manage it comfortably.

RatGit, 2019-05-18:
I meant Odroid XU4, (also 30 fps is probably a bit ambitious)

alphanumeric007, 2019-05-19:
I may try a lower frames per second, and reduce the movie quality. I just wish the Pi wouldn't lock up completely. I can't even go in and turn movies off. Plus if I power cycle it, it immediately starts recording again, or try's to.