motioneye-project/motioneyeos #1924: Cloud cleanup not working

patrickhener, 2019-05-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi 3B

### Camera

I am using the following type of camera: Raspberry Pi Camera
My camera model is: v2 Noir

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet
### Peripherals

- 

### Log Files

I consider the following log files relevant to this issue:
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3197233/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3197234/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3197235/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3197236/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3197237/motion.log)

### Screenshots
[image]
[image]
 
# Issue

So I configured cloud cleanup to cleanup with the files on gdrive. But it just doesn't.

ccrisan, 2019-06-03:
Judging from your logs, the given folder does not exist on Google Drive:

    Exception: folder with name "/Ueberwachung" does not exist

patrickhener, 2019-06-03:
That's what I thought at first, too. But it does. It is there. It does upload to that folder correct. It just does not cleanup afterwards.

bob-lee, 2019-06-11:
@patrickhener I am using gdrive,  `Loacation` field configured without leading slash and curious if that's causing your problem. I will test on my rpi and see if leading slash reproduces the problem.

bob-lee, 2019-06-12:
I could reproduce this issue with Location '/CCTV2' so current code doesn't handle a leading slash properly. I will prepare the fix in `motioneye` project.
@ccrisan In `clean_cloud` feature, I assumed that a folder named in `Location` field is a direct child of the 'root' so it would not like `'A/B'`. Is this assumption valid?

ccrisan, 2019-06-16:
https://github.com/ccrisan/motioneye/pull/1310 has been merged and the next motionEyeOS nightly build will have it.

closed 2019-06-16 by ccrisan

patrickhener, 2019-07-01:
I am on dev20190620 now and still the cloud cleanup is not working. Neither with or without a leading slash.

bob-lee, 2019-07-01:
@patrickhener can you please provide logs without a leading slash?

patrickhener, 2019-07-02:
@bob-lee : My bad! Actually it does work now. Sorry for the inconveniences.