motioneye-project/motioneyeos #1929: Problems in arm/disarm motion detection via motion Web API

SebastianSchultz, 2019-05-26:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
motionEyeOS 20190427

### Board Model
Raspberry Pi 3 B+

### Camera
2 Network cameras Wansview W3

### Network Connection
Ethernet
### Peripherals

I am using the following peripherals that I consider relevant to this issue:
None

### Log Files
I consider the following log files relevant to this issue:
[motion.log](https://github.com/ccrisan/motioneyeos/files/3221083/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3221084/motioneye.log)

Hello,

I have a problem with arm/disarming the motion detection via Motion Web API.
I pause the motion detection by the following urls:

http://192.168.178.245:7999/1/detection/pause
http://192.168.178.245:7999/2/detection/pause 

with the following urls I can check the status of the motion detection:

http://192.168.178.245:7999/1/detection/status
http://192.168.178.245:7999/2/detection/status

That works fine but after a few minutes without any changes at the web interface or something the detection resumes.
The working schedule is off for both cameras.

I need this option to activate and deactivate the motion detection via URL from another system.

Can you please help whats going on here?

SebastianSchultz, 2019-06-11:
Noone has an idea?

jasaw, 2019-06-12:
Run motion in the foreground with debug logging, see what's happening in the few minutes that causes it to resume motion detection.
```
/etc/init.d/S85motioneye stop
cd /data/etc
motion -c motion.conf -d 6
```

If it's a motion issue, report it to motion developers: https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue

SebastianSchultz, 2019-06-12:
Hello jasaw,

I stopped motioneye via

`sudo meyectl stopserver -b -c /path/to/motioneye.conf`
'sudo service motion stop'

then I started motion manually by 

`sudo motion -c motion.conf -d 6`

Then I stopped the detection via URL on both of my cameras:

http://192.168.178.245:7999/1/detection/pause
http://192.168.178.245:7999/2/detection/pause

Now the detection is still remaining on "Pause".

Here the Motion log of the foreground run:

`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [STR] webu_start_strm: Starting camera 1 stream on port 8081
[0:motion] [NTC] [STR] webu_strm_ntc: Started camera 1 stream on port 8081
[0:motion] [NTC] [STR] webu_start_strm: Starting camera 2 stream on port 8082
[0:motion] [NTC] [STR] webu_strm_ntc: Started camera 2 stream on port 8082
[0:motion] [NTC] [STR] webu_start_ctrl: Starting webcontrol on port 8080
[0:motion] [NTC] [STR] webu_start_ctrl: Started webcontrol on port 8080
[0:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 57.64.101 libavformat version 57.56.101
[0:motion] [NTC] [ALL] translate_init: Sprache: Deutsch
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 1 is from camera-1.conf
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 1 Camera Name: Eingang Service: rtsp:
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 2 is from camera-2.conf
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 2 Camera Name: Carport Service: rtsp:
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 867
[2:ml2:Carport] [NTC] [ALL] motion_init: Camera 2 started: motion detection Enabled
[2:ml2:Carport] [NTC] [VID] vid_start: Opening Netcam RTSP
[1:ml1:Eingang] [NTC] [ALL] motion_init: Camera 1 started: motion detection Enabled
[1:ml1:Eingang] [NTC] [VID] vid_start: Opening Netcam RTSP
[2:ml2:Carport] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Carport) connected
[2:ml2:Carport] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[3:nc3:Carport] [NTC] [NET] netcam_rtsp_handler: Normal resolution: Camera handler thread [3] started
[1:ml1:Eingang] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Eingang) connected
[1:ml1:Eingang] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[4:nc4:Eingang] [NTC] [NET] netcam_rtsp_handler: Normal resolution: Camera handler thread [4] started
[4:nc4:Eingang] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Eingang) connected
[1:ml1:Eingang] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 55 items
[3:nc3:Carport] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (Carport) connected
[2:ml2:Carport] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 55 items
`

When I'm now stopping motion and starting the Motioneye via 

`sudo meyectl startserver -b -c /path/to/motioneye.conf`

and set the cameras to detection pause then they also are remaining in pause.

But if I reboot the Raspi so Motioneye and motion are running in daemon mode on startup and set the cameras to pause then the problem occurs again that the cameras are only a few minutes (4 minutes and 30 seconds first measure, 4 minutes 46 seconds at the second measure) in pause mode and then they switch to active mode.

I hope this helps??? What I'm doing wrong?

SebastianSchultz, 2019-06-18:
Any ideas on this?

jasaw, 2019-06-20:
I would enable debug logging, reboot the device, and pause the detection. Wait for them to switch back to active detection mode, then grab the log files (motion.log, motioneye.log) and attach them here.

closed 2019-06-28 by SebastianSchultz