motioneye-project/motioneyeos #1931: Motioneyeos (Motion detect on or off via Switch on GPIO)

Schintgen, 2019-05-28:
Hello, I have Motioneyeos installed and can not find out how I can disable or enable motion detection with a switch on RPI GPIO . Is there an example for this? Yours sincerely.

pdisner, 2019-06-01:
Id love to know too. I think that youll need to use the motioneye part of the program, but i have no idea really. Id love to have sensors. I cheated and bout a Pioneer 600 Raspberry Pi had that has a ton of built in sensors  that I have hooked up to a Raspberry Pi A+. I figured that i could make somethng work without all the sensor library hunting.

burgerlX, 2019-06-18:
I would loooooooooove to be able do such things. eg. Start recording when a GPIO input gets triggered (motion sensors, laser/light sensors...) or be able to add gpio microphones and lights

jasaw, 2019-06-20:
Something like this that polls the GPIO pin (assuming GPIO high value to enable motion detection). You run this as a standalone program. Run this program from `/data/etc/userinit.sh`.
Change the `DET_EN_GPIO_PIN` to whichever pin you have connected to your switch.
Change the background flag to `True` (note the upper case T) if you want it to daemonize (needed if running from `userinit.sh`).

```python
#!/usr/bin/python
""" Watches GPIO pin for enable/disable motion detection """

import sys
import os
import signal
import time
import cStringIO
import pycurl
import RPi.GPIO as GPIO

terminate = False
DET_EN_GPIO_PIN = 17

def handle_signals(signum, stack):
    global terminate
    if signum == signal.SIGTERM or signum == signal.SIGINT:
        terminate = True

def createDaemon():
    UMASK = 0
    WORKDIR = "/"
    MAXFD = 1024
    REDIRECT_TO = "/dev/null"
    if (hasattr(os, "devnull")):
        REDIRECT_TO = os.devnull

    try:
        pid = os.fork()
    except OSError, e:
        raise Exception, "%s [%d]" % (e.strerror, e.errno)
    if (pid == 0):
        os.setsid()
        try:
            pid = os.fork()
        except OSError, e:
            raise Exception, "%s [%d]" % (e.strerror, e.errno)

        if (pid == 0):
            os.chdir(WORKDIR)
            os.umask(UMASK)
        else:
            os._exit(0)
    else:
        os._exit(0)

    import resource
    maxfd = resource.getrlimit(resource.RLIMIT_NOFILE)[1]
    if (maxfd == resource.RLIM_INFINITY):
        maxfd = MAXFD
    for fd in range(0, maxfd):
        try:
            os.close(fd)
        except OSError:
            pass
    os.open(REDIRECT_TO, os.O_RDWR)
    os.dup2(0, 1)
    os.dup2(0, 2)
    return(0)

def http_req(url):
    curl = pycurl.Curl()
    curl.setopt(pycurl.URL, url)
    response = cStringIO.StringIO()
    curl.setopt(pycurl.WRITEFUNCTION, response.write)
    response_headers = []
    curl.setopt(pycurl.HEADERFUNCTION, response_headers.append)
    try:
        curl.perform()
    except pycurl.error, e:
        print str(e)
    response_code = curl.getinfo(pycurl.RESPONSE_CODE)
    response_type = curl.getinfo(pycurl.CONTENT_TYPE)
    curl.close()
    response_data = response.getvalue()
    response.close()
    return response_code

def run():
    global terminate
    background = False

    # TODO: Get 'daemonize' and 'switch gpio pin' options from command line arguments

    if background:
        createDaemon()

    signal.signal(signal.SIGTERM, handle_signals)
    signal.signal(signal.SIGINT, handle_signals)

    GPIO.setmode(GPIO.BCM)
    GPIO.setwarnings(False)
    GPIO.setup(DET_EN_GPIO_PIN, GPIO.IN,pull_up_down = GPIO.PUD_UP)

    GPIO.input(DET_EN_GPIO_PIN)

    prev_state = None
    while not terminate:
        time.sleep(0.5)
        cur_state = GPIO.input(DET_EN_GPIO_PIN)
        if cur_state == False:
            url = 'http://localhost:7999/1/detection/pause'
        elif cur_state == True:
            url = 'http://localhost:7999/1/detection/start'
        if prev_state != cur_state:
            rc = http_req(url)
            print rc
            prev_state = cur_state

if __name__ == "__main__":
    run()
```

**DISCLAIMER:** I haven't tested this code, use at your own risk.

If you have more than 1 camera, change the `camera_index` accordingly in the URL: `http://localhost:7999/<camera_index>/detection/pause`

@burgerlX What you want is here: https://github.com/ccrisan/motioneyeos/issues/842#issuecomment-414375686

aleraco, 2021-02-12:
it doesn't work to me