motioneye-project/motioneyeos #1932: Connecting Motion Eye to Hotel Wifi

ricso, 2019-05-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ricso, 2019-05-28:
Is there a way to use the Raspbian GUI to connect to WiFi manually before running Motion Eye OS? I saw a post where the user setup MotionEye OS on top of Raspbian and did that but there are no instructions on how to do it.  I am trying to connect my PiZero with Motion Eye OS to a hotel wifi, which needs to be done manually, since I need to enter my room number and last name before I can connect

Chiny91, 2019-05-29:
Both Raspbian and motioneyeos are operating systems (OSs); you can't have both running on one R Pi Zero.   You probably saw a post describing the installation of motioneye (a web front end to motion) on Raspbian - this may well do what you want.   You could look at the motioneye (not motioneyeos) [Github pages](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian).   I have such a system running (not on an R Pi) and it was simple to get going - give it a try.   Google will then give pointers to lots of help re connecting Raspbian to hotel wifi.

ricso, 2019-05-30:
> Both Raspbian and motioneyeos are operating systems (OSs); you can't have both running on one R Pi Zero. You probably saw a post describing the installation of motioneye (a web front end to motion) [...]

Thank you for your comment.  I also believed that installing MotionEye on Raspbian would solve my problems, so I did that.  I can now connect to a Hotel Wifi (or any wifi, for that matter, since I have the Raspbian GUI) but I can't get Motion Detection to work and the python script that sends me a Telegram message on motion also dos not work (it runs perfectly on my other Pi with MotionEye OS).  I suspect it's related to MotionEye being run by root at boot but I don't know how to solve it!  Any help on this will be greatly appreciated!

ricso, 2019-06-10:
I am closing this issue, since my original problem was not being able to connect to a hotel wifi and that has been resolved by installing MotionEye over Raspbian.

closed 2019-06-10 by ricso