motioneye-project/motioneyeos #1934: Request: How to use motioneye without motioneyeos

Gualberto-Vannini, 2019-05-29:
Hi folks. 

I have a raspberry with jessie as SO. I am using it to doing many stuff as: 
- Temperature
- Umidity
- Access Point
- Send data to db
- audio speaker 
...so on.

I would like to add the motioneye system however I don't want to start from the motioneyeos and then reply all my system. I would like to install all the motioneye inside my SO. 

Any suggestion if there is some package do do it or if....it will exist? 

Thank you

pdisner, 2019-06-01:
sounds like a cool project. In fact, something that i thought of a while ago, but couldnt find an easy wasy to do it. mayb the best easy is to have seperate devices

ccrisan, 2019-06-01:
Sure, check out [motionEye](https://github.com/ccrisan/motioneye).

closed 2019-06-01 by ccrisan

Gualberto-Vannini, 2019-06-03:
> Sure, check out [motionEye](https://github.com/ccrisan/motioneye).

wow! I will post my updates. 
thank you for your attentions!