motioneye-project/motioneyeos #1936: QuickFix Documentation: Tweaks - Cron jobs syntax

Asharas, 2019-06-02:
Very quick correction on the wiki (I'm new to GH and haven't find how to contribute to the wiki if it's possible).

In [Running Commands At Regular Intervals (Cron Support)](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support)

Example given to run command every 5 min is: 
```
3,8,13,18,23,28,33,38,43,48,53,58 * * * * sleep 20 ; curl -S "http://freedns.afraid.org/dynamic/update.php?Rry68SBhp93GbgzL7T57FEb6kK9djtvjXGTDEADBEEF=" >> /var/log/dyndns.log 2>&1 &
```

A cleaner version should be:
```
3-59/5 * * * * sleep 20 ; curl -S "http://freedns.afraid.org/dynamic/update.php?Rry68SBhp93GbgzL7T57FEb6kK9djtvjXGTDEADBEEF=" >> /var/log/dyndns.log 2>&1 &
```
`3-59` is meant to keep the offset from the example, see this [StackOverflow post](https://stackoverflow.com/questions/12786410/run-cron-job-every-n-minutes-plus-offset)
`/5` is "run every 5th min" (in the interval specified just previously, see linked tool below).  

Neat tool I discovered while looking for documentation to support the issue:  
[Crontab Guru](https://crontab.guru/#3-59/5_*_*_*_*  )

ccrisan, 2019-06-11:
Have you actually tested the "/" syntax on motionEyeOS? I'm asking this because we're using `busybox` crond functionality and `busybox` applets are often missing various less-popular features.

Asharas, 2019-06-11:
I actually hadn't tested prior to opening the issue, I should have, my bad (I assumed `crond` implementation was standard).  

I have now and it works.  

I'll be more careful and test before next time.

ccrisan, 2019-06-15:
Article updated. Thanks!

closed 2019-06-15 by ccrisan