motioneye-project/motioneyeos #1939: Raspberry Pi 3 b+ fails to set up wired connection, panic reboots for hours

kc10guru, 2019-06-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

kc10guru, 2019-06-05:
First time using the software, burned SD card and booted device, gets stuck at configuring wired network, says there is no link.  I have disconnected the Ethernet cable and connected it to my laptop and verified the connection works. Tried for 1.5 hours and it constantly reboots and extends the reboot by 20 seconds each time. now at 280 seconds. Using the latest software and new Raspberry Pi 3B+

dritonkb, 2019-06-05:
Hi, Make sure the ethernet cable or eth-port from your router is working. You can also look at the ethernet port on your Raspberry Pi 3b+ if it lights up when you plugg it in.. 

BS
Driton K

kc10guru, 2019-06-05:
I figured it out last night.  I was using a crossover cable by mistake.
When plugged into a laptop it worked but when connected to the raspberry pi
it would pull an IP address.   Changed cable and now it’s working great.

kc10guru, 2019-06-05:
Found the problem last night, apparently I was using a crossover cable for the Ethernet connection. Oops, changed the cable and the box pulled the IP address. Go figure.

closed 2019-06-05 by kc10guru