motioneye-project/motioneyeos #1941: Very slow response to WPA supplicant

SupraGuy, 2019-06-06:
Motioneye OS 20190427
Raspberry Pi 3 Model A+ using built-in WiFi
32G Micro SD card
Raspberry Pi camera

Motioneye OS eventually connects to the wireless network, but takes several minutes to do so. Does not seem to matter 2.4G or 5G. No special characters in SSID nor WPA passphrase, SSID is visible.

SupraGuy, 2019-06-07:
If I plug in the USB modules that I used with my old Pi 1 model B boards, it finds them quickly, but cannot connect to 5G. Not surprising since the USB adapters are only wireless n.

nelsogar, 2019-06-07:
Motioneyeos 20190427
RPi 3 B+, RPi Camera

I am having the same issue. If I try to use wireless, the boot fails on WPA_Supplicant, and goes into a panic reboot. I did not let it continue to panic and reboot for several minutes to see if it eventually would work, but I will give that a try. I also noticed if I am connected with an ethernet cable and try to set a static address, the eth0 will fail, and I will again go into the panic, reboot cycle.

nelsogar, 2019-06-07:
As a follow-up, on a fresh, newly written image of the current version, with the only change from defaults being turning on wifi and pulling the ethernet cable, it has been failing and doing panic reboots for over an hour. I don't think it's going to magically fix whatever issue it is having with wpa_supplicant on its own. Oh, and I did make sure I used the image for an RPi 3.