motioneye-project/motioneyeos #1942: IP Address Not Visible

FuriousLogic, 2019-06-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: N/A

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

I can see the Pi has connected to the wifi by having a monitor connected.  Plus I can see the device on my router's admin page.
However, the IP address does not show up on Angry Scanner nor can I browse to it (get a "This site can’t be reached" message)

jasaw, 2019-06-10:
Please provide more information about your problem.
Your Pi has connected to wifi. You verified this by looking at a monitor attached to the Pi. I presume the Pi managed to get an IP address from your DHCP server. Have you tried pinging your router and Angry Scanner machine from this Pi? What is the IP address and subnet mask of this Pi?
On your Angry Scanner machine, have you tried pinging the IP address of the Pi? What is the IP address and subnet mask of this machine?

FuriousLogic, 2019-06-11:
Hi,
Firstly, thanks for getting back to me!

Here's the bullet points:
- I connected a monitor to observe the boot. It connected to the WiFi &
allocated an ip address.
- I could connect to my router & see the pi connected on the same address.
- Angry Scanner, on a PC on the same WiFi network, could not see anything
connected on that address.
- needless to say, if I tried to browse the ip from that PC it couldn't see
anything there.

I will try pinging from the pi & check the subnet masks!

Thanks!
Barry

DanTheOrange, 2019-06-25:
I am having this exact same issue.

- I've connected to a monitor and it's booted successfully, giving me a prompt to go to my browser and an IP.
- I'm able to see my Pi in the routers admin page.
- Angry scanner does not pick it up and I can't ping it.

When I go to the IP in the browser or I try to ping it I don't get a straight failure. If I go to an IP that isn't in use it bails out and tells me it doesn't exist. Instead it will try to load in the browser window and eventually time out. Or while pinging do a similar thing. I'm not near it at the moment to provide more info, but I am happy to later on.

FuriousLogic, 2019-06-25:
Hey Dan,
In short I gave up for the day.  Booted it up the following day & all was fine...  Go figure.

I wonder if the IP address is restricted for some reason somewhere in the network.  I had the problem with 192.168.1.13, then it worked with 192.168.1.19.  May be a red herring...  Sorry I can't be of more help.
Barry

DanTheOrange, 2019-06-25:
Okay, so it works for me now too. But my Journey was repeatable three times. 

I got home today and gave it another go, starting from scratch. I flashed the SD card and got the same IP as last night.
[image]

I then tested to see if it would work. With no luck! It had an IP address, but it wasn't found by ping or Angry IP Scanner.
[image]
[image]
[image]
[image]

But, this is the consistent and repeatable part, after the first time this happened I did it two more times to confirm and I had the same issue on both occasions. On the Pi I logged in at the prompt I had there, from this point I was able to access it via the webpage at the IP and ping my PC. From the instructions given as this is meant to be headless I didn't think I would have to do this.
[image]

I'm using version `motioneyeos-raspberrypi-20190427.img.xz`
I used balenaEtcher to put the image on the sd card
I used sublime with EOL unix to make the `wpa_supplicant.conf`
Using no hardware and the pi plugged into a portable power pack at 5v 2A

DanTheOrange, 2019-06-25:
I could add that every time I turn it on I have to wait for the Pi to prompt a login in the terminal on the Pi and then actually log in on it or I can't use it via the web interface. This means that if the power drops it ceases working for its intended purpose. This seems to be a consistent thing. I've turned it on and off half a dozen times now. I'm just about to test it on another Pi zero w and I'll look on another Pi I have somewhere (a 1A?) if I can find it.

jasaw, 2019-06-26:
@DanTheOrange Do you have issues pinging other wifi devices from your PC? The Pi can reach other devices on your network, but your PC can't ping your Pi. This sounds like a routing issue on your PC or router.

DanielLewisArm, 2019-06-26:
I have no issues anywhere else in my network.

Everything works perfectly fine as soon as I log in on the terminal on the Pi. It's this part here, it's not working. Then as soon as I type in "admin" with no password it works completely.
[image]

DanTheOrange, 2019-06-26:
I have no issues anywhere else in my network.

Everything works perfectly fine as soon as I log in on the terminal on the Pi. It's this part here, it's not working. Then as soon as I type in "admin" with no password it works completely.
[image]

DanTheOrange, 2019-06-26:
Sorry, my explanation might not be good. I was following the instructions exactly. As soo as I get to this point in the instructions I have the issue.

>As soon as your motionEyeOS is ready it will listen on port 80 and present you a web user interface. To access it, enter the IP address of your board in the address bar of your browser. To find the address, connect to your router (most routers can display the addresses of all connected devices) and look for a device name beginning with "meye-". Alternatively, you can use a network scanner such as the free mobile app Fing.

The problem is, at this point I can't ping the Pi or go to the IP in the browser. But if I plug in a screen and keyboard I am presented with the prompt `meye-590afe9f login:`. At this point I use the credentials the instructions give `admin` and no password. This for some reason allows me to then ping it or go and view it in the browser and everything works as expected.

closed 2020-11-10 by FuriousLogic

CTHRX, 2024-07-17:
i have same issue in this problem can someone help me