motioneye-project/motioneyeos #1944: How connect rasp to tv with hdmi

mb39924, 2019-06-08:
I want to view my steams on a tv using  hdmi. Plugged it in and it doesn't work. I thought there used to be a setting to turn hdmi on but I can't find it anywhere. 

Can anyone point me in the right direction?

auxz2jz, 2019-06-08:
Turn on Fast network mode and there's an HDMI preview works great for me just trying to get the rstp to work now

mb39924, 2019-06-09:
Ah ok.. Im using a raspi with motioneyeos as a hub with mom-motioneyeos cameras. So no fast network mode for me.

Would be a great feature to be able to connect a monitor to a hub.

ShibumiDesign, 2019-06-10:
If you have a smart tv connected to your local network with a web browser in your tv (many newer ones do), just browse to your MotionEyeOS hub IP address... or connect a laptop or pc to your tv and browse to your hub IP address

ccrisan, 2019-06-10:
@mb39924 motionEyeOS is not designed to be used as a frontend. You should use a dedicated PC with a web browser.

closed 2019-06-10 by ccrisan