motioneye-project/motioneyeos #1945: This software is a piece of shit!

Kabuse, 2019-06-08:
I am a software developer since 25 years, but I have never seen such an piece of shit like this! Tried to make it run on a RasPi Zero W. Absolute frustrating to say the least. Can connect one time after writing the image, try to change some settings, nothing works. Crashing, error messages and unable to connect. See the live video for 10s, then it drops. Can not log in anymore until I reflash the image again. Wasted about six hours today. What a pice of shit! Anybody who want to use this with a RasPi Zero W, don't waste your time on this fucking piece of shit software! I wish, I would have never tried! One of the most frustrating moments in my It-life!

mb39924, 2019-06-08:
I have a raspi zero w camera working without any problems. At the moment as a fast network camera connected to a central hub (raspi 3b + with motioneyeos)

I've had some problems with a raspi zero cam as a stand alone system but that was mostly because of the wifi capabilities of the raspi and storage issues. No problems related to the motioneyeos software.

ncguk, 2019-06-08:
I stopped using motioneyeos on a Pi Zero W because I would get frequent camera drop-outs, freezes etc. That doesn't mean motioneyeos a piece of shit. It works fine on a Pi 3. The Pi Zero W is just not powerful enough to handle motioneyeos in some situations.

mb39924, 2019-06-09:
Btw, for someone who is probably over 40 a title like "This software is a piece of shit!" is kinda childish. Take a breathe. Just because it isn't doing what you where hoping it would do, doesn't mean the software is a piece of shit. As a software developer you must have checked the error logs. What did it say?

ccrisan, 2019-06-09:
@Kabuse I always enjoy this kind of constructive criticism. Keep it coming. Or is this all you've got?

MrJBSwe, 2019-06-09:
"Kabuse doesn’t have any public repositories yet. " ..  just saying ;-)

Kabuse, 2019-06-09:
I have to apologise for beeing a bit impulsive. It was quite late and very frustrating.

Anyway, I was able to identify that it was a hardware issue. Seems that newer RasPi Zero Ws do not work. I stumbled up on this:

https://github.com/raspberrypi/linux/issues/2555#issuecomment-394105250

I have quite a few Raspberries here working and was able to find one Zero W of the old revision. After switching the devices and using modest overclocking, everything seems to work just fine, at least for the last couple of hours.

Don't know what is the exact problem of the new revisions of the Zero Ws, but as far as I can tell, they will not work with MotionEyeOS. I als tried to use use Motion with the new revision board and ran into the same Kernel error like in the thread mentiond above.

ccrisan, 2019-06-09:
It might be worth considering a downclock to 950 MHz by default then.

ccrisan, 2019-07-15:
Could you please try the latest [nightly dev build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and see if the WiFi problem persists?