motioneye-project/motioneyeos #1946: No Cams load after dev20190607 update

ShibumiDesign, 2019-06-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20190607

### Board Model

I am using the following board/model: 1x Raspberry PI 3B+, and 2x Raspberry Pi Zero W).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: 1x Raspberry Pi NoIR Cam, 2x Waveshare WideAngle Pi Cams.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
So after having issues with 2 Pi Zero W cams that keep freezing and/or dropping randomly, I updated all Pi's to latest dev version dev20190607, motioneyeos loads, but sits in a state of flux never loading the streams on each of the cameras. The menu button activates the menu, but is all greyed out and nothing is selectable. Dropdown to select camera is empty and cannot even add a camera, and delete button says no camera to remove.

I even tried manually rebooting each of the 3 systems, and still same result.

tigernero79, 2019-06-10:
same my problem but this happens only with Chrome browser also for smartphone.  all ok with Firefox browser

ShibumiDesign, 2019-06-10:
Just confirm, works in safari, not it Chrome (latest version - Version 75.0.3770.80 (Official Build) (64-bit)) on Mac OSX High Serra

ccrisan, 2019-06-10:
Clear the browser's cache. The javascript code has been changed.

tigernero79, 2019-06-10:
the cache had to be reset, done with ccleaner. or you can do it from chrome itself. thank you it worked for me.

closed 2019-06-10 by ccrisan

potto, 2019-08-07:
Hi, can you please check if MotionEyeOS **does work on Chrome** 76.0.3809.100 64bit slovak language? I have no stream from cameras. **It works normally on Edge**. Thank you

motionEye Version | 0.40
Motion Version | 4.2.2
OS Version | motionEyeOS dev20190728

Cache was removed, but the issue persists.