motioneye-project/motioneyeos #1948: Build from source with nginx fails

gilsonmandalogo, 2019-06-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry PI 2 Model B

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

I can build from source on `master` or `dev` successfully, but when I try to add nginx as target package, both branches fails.
Before the latest update on `master` branch, I had build successfully with nginx additional target package

Note: All builds are done using ubuntu inside docker container.

The error occurs on the `Generating common rootfs tarball` step.

Bellow are my latest tty outputs:
```
>>>   Generating common rootfs tarball
rm -rf /home/motioneyeos/output/raspberrypi2/build/buildroot-fs
mkdir -p /home/motioneyeos/output/raspberrypi2/build/buildroot-fs
rsync -auH /home/motioneyeos/output/raspberrypi2/target/ /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target
echo '#!/bin/sh' > /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
echo "set -e" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
echo "chown -h -R 0:0 /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
printf '   	- - input -1 * - - - Input device group\n 	sshd -1 sshd -1 * /var/empty - - SSH drop priv user\n 	ftp -1 ftp -1 * /home/ftp - - Anonymous FTP User\n\n' >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/users_table.txt
PATH="/home/motioneyeos/output/raspberrypi2/host/bin:/home/motioneyeos/output/raspberrypi2/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/motioneyeos/support/scripts/mkusers /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/users_table.txt /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
cat system/device_table.txt > /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/device_table.txt
printf '   	/bin/busybox                     f 4755 0  0 - - - - -\n 	/var/lib/nginx d 755 33 33 - - - - -\n 	/var/empty d 755 root root - - - - -\n\n' >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/device_table.txt
echo "/home/motioneyeos/output/raspberrypi2/host/bin/makedevs -d /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/device_table.txt /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
printf '   	tar cf /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/rootfs.common.tar --numeric-owner --exclude=THIS_IS_NOT_YOUR_ROOT_FILESYSTEM -C /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target .\n' >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
chmod a+x /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
PATH="/home/motioneyeos/output/raspberrypi2/host/bin:/home/motioneyeos/output/raspberrypi2/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/motioneyeos/output/raspberrypi2/host/bin/fakeroot -- /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/fakeroot.fs
rootdir=/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target
table='/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/device_table.txt'
makedevs: Cannot create directory `/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target/var/lib/': No such file or directory
makedevs: line 22: chown failed for /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/target/var/lib/: No such file or directory
fs/common.mk:71: recipe for target '/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/rootfs.common.tar' failed
make: *** [/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/rootfs.common.tar] Error 1

root@6d3bd08590b5:/home/motioneyeos# ls -la output/raspberrypi2/build/buildroot-fs/target/var/
total 16
drwxr-xr-x  4 root     root     4096 Jun 11 10:31 .
drwxr-xr-x 20 root     root     4096 Jun 11 10:31 ..
lrwxrwxrwx  1 root     root        4 Jun 11 10:31 cache -> /tmp
drwxr-xr-x  2 root     root     4096 Jun 11 09:35 empty
lrwxrwxrwx  1 root     root       12 Jun 11 10:31 lib -> /data/varlib
lrwxrwxrwx  1 root     root        4 Jun 11 10:31 lock -> /tmp
lrwxrwxrwx  1 root     root        9 Jun 11 10:31 log -> /data/log
lrwxrwxrwx  1 root     root        4 Jun 11 10:31 run -> /tmp
lrwxrwxrwx  1 root     root        4 Jun 11 10:31 spool -> /tmp
lrwxrwxrwx  1 root     root        4 Jun 11 10:31 tmp -> /tmp
drwxr-xr-x  2 www-data www-data 4096 Jun 11 10:31 www

root@6d3bd08590b5:/home/motioneyeos# git log   
commit 9dcb36a72b05647b30aeb0d598191455ef363cf1 (grafted, HEAD -> master, tag: 20190427, origin/master)
Author: Calin Crisan <ccrisan@gmail.com>
Date:   Sat Apr 27 21:35:38 2019 +0300

    motionEye: update to 0.40
```

ccrisan, 2019-06-16:
Can you please try building the latest dev?

gilsonmandalogo, 2019-06-17:
@ccrisan Thanks for your help :)

I tried now on the latest dev branch, and a clean build. The build fails on the same step, but with a slightly different error.

Log:
```
>>>   Generating filesystem image rootfs.tar
mkdir -p /home/motioneyeos/output/raspberrypi2/images
rm -rf /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar
mkdir -p /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /home/motioneyeos/output/raspberrypi2/target/ /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target
echo '#!/bin/sh' > /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
echo "set -e" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
echo "chown -h -R 0:0 /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
PATH="/home/motioneyeos/output/raspberrypi2/host/bin:/home/motioneyeos/output/raspberrypi2/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/motioneyeos/support/scripts/mkusers /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/full_users_table.txt /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
echo "/home/motioneyeos/output/raspberrypi2/host/bin/makedevs -d /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/full_devices_table.txt /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target" >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
printf '   \n' >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
printf '   	(cd /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target; find -print0 | LC_ALL=C sort -z | tar  -cf /home/motioneyeos/output/raspberrypi2/images/rootfs.tar --null --xattrs-include='\''*'\'' --no-recursion -T - --numeric-owner)\n' >> /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
chmod a+x /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
PATH="/home/motioneyeos/output/raspberrypi2/host/bin:/home/motioneyeos/output/raspberrypi2/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" FAKEROOTDONTTRYCHOWN=1 /home/motioneyeos/output/raspberrypi2/host/bin/fakeroot -- /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/fakeroot
rootdir=/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target
table='/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/full_devices_table.txt'
makedevs: Cannot create directory `/home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target/var/lib/': No such file or directory
makedevs: line 3: chown failed for /home/motioneyeos/output/raspberrypi2/build/buildroot-fs/tar/target/var/lib/: No such file or directory
fs/tar/tar.mk:16: recipe for target '/home/motioneyeos/output/raspberrypi2/images/rootfs.tar' failed
make: *** [/home/motioneyeos/output/raspberrypi2/images/rootfs.tar] Error 1

root@c1cdf964ef49:/home/motioneyeos# git log
commit ba9b6d14a927e512b52e82572334ae44310bd1b4 (grafted, HEAD -> dev, origin/dev, origin/HEAD)
Author: Calin Crisan <ccrisan@gmail.com>
Date:   Sun Jun 16 21:29:07 2019 +0300

    motioneye: update to b65febd63fce1fd69563c6b8b299789ad0c40684
```

ccrisan, 2019-06-17:
Have you customized in any kind the raspberrypi2 defconfig?

gilsonmandalogo, 2019-06-17:
No, I don't customized the defconfig, I customized only target packages on `/output/raspberrypi2/.config` through `./build.sh raspberrypi2 nconfig` command, removing wifi stuffs, adding nginx (plus some submodules), and adding fail2ban.

I know the problem is nginx because I tried (not after this latest change on dev) all other changes, and without nginx the build works.

Here are my [/output/raspberrypi2/.config](https://pastebin.com/1DX8G4pU) file.

Thanks :)