motioneye-project/motioneyeos #1951: recordings are just still images - the entire video is just a single frame

MasterReboot, 2019-06-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20190427

### Board Model

I am using the following board/model:  Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: Logitech C270

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection: Wired

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. 
[boot.log](https://github.com/ccrisan/motioneyeos/files/3281734/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3281735/motioneye.log)

boot.log)

MasterReboot, 2019-06-12:
I have opened videos saved to my sdcard and also uploaded to google drive. I opened various videos spread out several different days, all with the same issue. I am using VLC to open the videos which are .mp4. I've played around with different settings, but nothing seems to have made a difference. Once I open the video, it's just a single frame throughout the entire recording.

jasaw, 2019-07-05:
@MasterReboot can you share the single frame video so we can see what the problem is?

MasterReboot, 2019-07-06:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/ccrisan/motioneyeos/issues/1951?email_source=notifications\u0026email_token=AMK2Z3CIGZ7WYC6CV7HEW5DP52KVVA5CNFSM4HXJQB4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZIJCWA#issuecomment-508596568",
> "url": "https://github.com/ccrisan/motioneyeos/issues/1951?email_source=notifications\u0026email_token=AMK2Z3CIGZ7WYC6CV7HEW5DP52KVVA5CNFSM4HXJQB4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZIJCWA#issuecomment-508596568",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Thank you for the response. The issue somehow corrected itselfSent from Samsung tablet