motioneye-project/motioneyeos #1952: Not able to attach Esp-32 cam

ashishbindal93, 2019-06-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: latest

### Board Model

I am using the following board/model: raspberry pi zero w

### Camera

I am using the following type of camera: Esp-32 based Simple MJPEG stream

My camera model is: none

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 Several Esp-32 based wifi cameras all connected to router.
[image]

The raspberry pi zero w is also connected via wifi to router.

When I try to add a camera 

Camera type : Simple MJPEG camera
URL : mjpeg url of the esp cam
(Eg: http://192.168.1.6:81/stream which works flawlessly on browser. Just open the url and streaming starts.)
Camera : No camera found

I don't understand why this fails to find camera.
 
Please help.

ashishbindal93, 2019-06-15:
Urls like http://192.168.1.5 work.

closed 2019-06-15 by ashishbindal93

ToXIc-Dev, 2019-08-26:
So did you manage to get this working and if so, how?

ashishbindal93, 2019-08-27:
Yes I got this working by using the "Add Network Camera" option. Initially it was not working but after a while it started to work. 
It detected the camera by either of the urls

1. http://192.168.1.6:81/stream      
2. http://192.168.1.6:81/stream/1

fjf-fernandes, 2019-10-21:
Hi @ashishbindal93 , when I try to add my ESP32 cam as network camera the same way you did I get the error "Empty reply from server".

Could you share the steps you took to flash and config the ESP32 cams?

Domy5, 2020-02-11:
> Yes I got this working by using the "Add Network Camera" option. Initially it was not working but after a while it started to work.
> It detected the camera by either of the urls
> 
> 1. http://192.168.1.6:81/stream
> 2. http://192.168.1.6:81/stream/1

ooo can you help us

sankarbha, 2020-02-12:
> Hi @ashishbindal93 , when I try to add my ESP32 cam as network camera the same way you did I get the error "Empty reply from server".
> 
> Could you share the steps you took to flash and config the ESP32 cams?

I used the sketch from [RNT](https://randomnerdtutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/) and added a new network camera with the http://ip-you-get-from-serial-debug-window. I worked perfectly.

The default sample CameraWebServer example does not stream when you hit the IP, one had to click the stream button to start streaming. I also tried @Domy5 's post like ip/stream or ip/stream/1 but still got the same "Empty reply from server".

Hope this would fix your issue.

ivanjf1978, 2020-09-23:
Hi everyone,

I have the same issue. I am investigate it is possible to connect the ESP32 Cam with the MotionEyeOS. As soon I will find the solution will let you know to everyone.

I trying to understand this information from this website

https://learn.circuit.rocks/esp32-cam-with-rtsp-video-streaming

Regards

mrlunk, 2021-02-10:
> Hi everyone,
>
> I have the same issue. I am investigate it is possible to connect the ESP32 Cam with the MotionEyeOS. As soon I will find the solution will let you know to everyone.
>
> I trying to [...]

Any luck yet ?