motioneye-project/motioneyeos #1954: unable to connect to Mobile Hotspot

ioangkin, 2019-06-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry PI 3 V1.2

### Camera

My camera model is: Standard Raspberry Pi  Camera V2.1

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

external monitor via HTMI cable

### Log Files

I consider the following log files relevant to this issue:

Unfortunately I cannot access the log files. I access the Raspberry Pi's memory card in my Windows 10 laptop and cannot see any log files, neither the var folder (/var/log/motioneye.log based on FAQ)

Issue:
Hi all, I cannot access the user interface via the web browser.

I am trying to connect the board (Raspberry Pi) with my phone's Mobile Hotspot but it doesn't seem to work. Based on the phone, the device is not connected, however the board doesn't restart due to luck of connection (panic restart).

Note that wpa_supplicant.conf and static_ip.conf are both found and read, as seen in attached.

Any advice is highly appreciated.

Thanks

ccrisan, 2019-06-17:
> however the board doesn't restart due to luck of connection (panic restart).

What do you mean by that? Does it or does it not restart?

> Note that wpa_supplicant.conf and static_ip.conf are both found and read

Most cases where users use static IP configurations, they fuck up the IP addresses and end up with a non-working IP configuration. Please try without a `static_ip.conf` and see if it works that way.

> ...as seen in attached.

There's nothing attached to this issue. Have you forgotten to attach some screenshots?

ioangkin, 2019-06-17:
Thank you for the prompt respond @ccrisan,

> > however the board doesn't restart due to luck of connection (panic restart).
> 
> What do you mean by that? Does it or does it not restart?

It does not restart. I would expect that it would restart due to no wifi found, however it does not. This indicates to me that as far as the board is concerned there is wifi connection, however the phone (acting as a router) does not see the connection.

> > Note that wpa_supplicant.conf and static_ip.conf are both found and read
> 
> Most cases where users use static IP configurations, they fuck up the IP addresses and end up with a non-working IP configuration. Please try without a `static_ip.conf` and see if it works that way.

Thanks, I will try without static ip configuration.. I just need to find how to reset the option as following boot the settings files are moved away from the boot folder and I only have access to that folder in windows 10.

> There's nothing attached to this issue. Have you forgotten to attach some screenshots?

Apologies, I forgot to attach.
[image]

ioangkin, 2019-06-18:
Update: I managed to reset static_ip configuration by just placing an empty (free of contents) static_ip.conf in the board's root folder.

And you were right @ccrisan, this resolved the issue. following, I was able to connect to the mobile hotspot.
The issue seems to be that the content was in capital letters and there were no quote marks around the address. Just for reference, the correct file looks like this:
```
static_ip.conf:
static_ip="192.168.43.56/24"
static_gw="192.168.43.1"
```

However, this has caused a new unexplained problem.
After connecting for few times with the raspberry Pi and the above static_ip.conf, initially on a router and then on my mobile hotspot (with different addresses in the static_ip.conf), both device's SSID stopped showing up on any wifi device (such as my laptop and the raspberry Pi), even after factory-resetting the router and restarting the phone. So practically now I cannot use my router and my phone's hot-spot to connect to the internet.

Any advice on this would be highly appreciated.

ioangkin, 2019-06-20:
@ccrisan Any ideas?

ccrisan, 2019-06-26:
The static IP settings are all uppercase. The fact that you wrote them in lowercase simply disabled them. There are two possible problems with your setup:
1. The gateway IP might not be that. In my case, my android AP's IP address is random.
2. You're missing the DNS server. If you're in doubt, use `8.8.8.8`.