motioneye-project/motioneyeos #1955: SD Card is being Corrupted

TamirSlo, 2019-06-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running MotionEyeOS version: 2019 (latest)

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: Generic Branded

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

N/A
 
# Issue Description:

Trying again to use the Operating System, referring to the issue of {“error”, ”internal server error”}, every time i use it, at some point of time, between several hours to several days, I get the error and the SD Card formatting is corrupted. Upon restarting, the Raspberry Pi cannot boot (attached a picture) and my laptop is not able to read the SD Card (says “no media” and that the capacity is 0GB).

This is not hardware related as i have used it before 24/7 for Website Hosting with no issues.

There are no logs as the SD Card is formatted / corrupted as the file system is unreadable.
[image]

ccrisan, 2019-06-17:
Have you tried it with a different SD card (brand/model)?

TamirSlo, 2019-06-17:
I have tried it with 3 different SD Cards.
SanDisk 32GB (class 10)
Kingston 32GB (class 6)
Generic Branded 16GB (class 10)

All three same outcome, with no previous warning. How long these lasted is random. As these file systems are corrupted I can no longer use these at the moment. I have only managed to recover (format) one SD Card using a camera that supports EXT file system.

With the one I recovered, i am now trying to install Raspbian Lite with MotionEye (only the web platform) and see how it goes.

Tamir

lukeswitz, 2019-06-22:
These cards burn out from heavy logging or not being shutdown properly. That’s how these boards work. Higher quality cards are less susceptible to this but it’s common among this community.

jasaw, 2019-06-26:
@lukeswitz You're talking about two different issues here: 
1. flash chip wearing after n write cycles. This is non-recoverable = buy a new SD card.
2. filesystem or data corruption caused by power loss. This is recoverable = flash your SDcard again or repair the filesystem.

To address your first issue:
If you are worried about wearing out the SD card, you could log to RAM instead of disk. I think /var/log is just a symlink to /data/log. You could change the symlink to point to /tmp instead.
You'll also need to store your media files somewhere, but you'll have to decide which storage technology you're comfortable with. Every storage tech has limit write(&read) cycles. Some SSDs have write cycles in the 4 digit range. Even magnetic spinning disks are specified to work up to some number of read&write cycles. Read the datasheets.

To address your 2nd issue:
MotionEyeOS mounts the rootfs and boot partitions as read-only by default, so power loss corruption is minimized. If you want zero chance of corruption, I suggest you store your media files in a separate disk/SDcard, and mount it as synchronous.

MarcoMagana, 2019-10-30:
Hi TamirSlo,

The same happened to me. I have not been hable to recover the memory card yet.

I’ll try with a DSLR camera to see if that helps.

andrsmllr, 2020-03-25:
Same happens for me with SD cards from Samsung (32GB), SanDisk (16GB) and Intenso (16GB). Used to work much better with older versions of MotionEyeOS, but unfortunately can't put my finger on the exact version.