motioneye-project/motioneyeos #1957: The time is wrong

burgerlX, 2019-06-18:
###Info
Version: the newest stable

Board: Raspberry pi 3B+

Cams: 1x Network Camera (tablet in a box)

Network: wired, tablet on 2.4G wifi

###Issue
The time is wrong
really really wrong

like 49 years wrong
the timestamp on the camera is 1970-1-1 02:10:00 (right now)

Anyone also got this issue?

burgerlX, 2019-06-18:
Looking at the log, this issue started when i removed a local mjpg camera.

log:
```
2019-06-18 20:07:08: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-06-18 20:07:18: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-06-18 20:07:28: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-06-18 20:07:29: [motioneye]     INFO: removing camera config file /data/etc/camera-1.conf...
2019-06-18 20:17:46: [motioneye]    ERROR: HTTP 404: Not Found (not found)
2019-06-18 20:17:46: [motioneye]  WARNING: 404 GET /nice%20ports%2C/Tri%6Eity.txt%2ebak (192.168.0.170) 2.51ms
2019-06-18 20:17:46: [motioneye]    ERROR: HTTP 404: Not Found (not found)
2019-06-18 20:17:46: [motioneye]  WARNING: 404 GET /nice%20ports%2C/Tri%6Eity.txt%2ebak (192.168.0.170) 2.40ms
2019-06-18 20:17:46: [motioneye]    ERROR: HTTP 405: Method Not Allowed
2019-06-18 20:17:46: [motioneye]  WARNING: 405 OPTIONS / (192.168.0.170) 2.31ms
2019-06-18 20:17:46: [motioneye]    ERROR: HTTP 405: Method Not Allowed
2019-06-18 20:17:46: [motioneye]  WARNING: 405 OPTIONS / (192.168.0.170) 2.24ms
2019-06-18 20:18:32: [motioneye]  WARNING: 403 GET /login/?_=1560881916086 (192.168.0.170) 2.43ms
2019-06-18 20:18:35: [motioneye]    ERROR: authentication failed for user admin
2019-06-18 20:18:35: [motioneye]  WARNING: 403 GET /login/?_=1560881918987&_username=admin&_login=true&_signature=fa94240076e7d382b098836b13d6972bd36546fa (192.168.0.170) 3.26ms
2019-06-18 20:18:41: [motioneye]    ERROR: authentication failed for user admin
2019-06-18 20:18:41: [motioneye]  WARNING: 403 GET /login/?_=1560881924334&_username=admin&_login=true&_signature=e57036dcd2e84fd3e05303169448f2edbefeff69 (192.168.0.170) 4.40ms
2019-06-18 20:19:24: [motioneye]     INFO: rebooting
2019-06-18 20:19:24: [motioneye]     INFO: executing "/sbin/reboot"
2019-06-18 20:19:24: [motioneye]     INFO: interrupt signal received, shutting down...
2019-06-18 20:19:24: [motioneye]     INFO: server stopped
2019-06-18 20:19:24: [motioneye]     INFO: tasks stopped
2019-06-18 20:19:26: [motioneye]     INFO: motion stopped
2019-06-18 20:19:26: [motioneye]     INFO: smb mounts stopped
2019-06-18 20:19:26: [motioneye]     INFO: bye!
1970-01-01 01:00:28: [motioneye]     INFO: hello! this is motionEye server 0.40
1970-01-01 01:00:32: [motioneye]     INFO: cleanup started
1970-01-01 01:00:32: [motioneye]     INFO: wsswitch started
1970-01-01 01:00:32: [motioneye]     INFO: tasks started
1970-01-01 01:00:32: [motioneye]     INFO: mjpg client garbage collector started
1970-01-01 01:00:32: [motioneye]     INFO: smb mounts started
1970-01-01 01:00:32: [motioneye]     INFO: server started
1970-01-01 01:00:29: [motioneye]     INFO: hello! this is motionEye server 0.40
1970-01-01 01:00:33: [motioneye]     INFO: cleanup started
1970-01-01 01:00:33: [motioneye]     INFO: wsswitch started
1970-01-01 01:00:33: [motioneye]     INFO: tasks started
1970-01-01 01:00:33: [motioneye]     INFO: mjpg client garbage collector started
1970-01-01 01:00:33: [motioneye]     INFO: smb mounts started
```

harisqazi1, 2019-06-19:
I don't have this issue, but have you tried using the timezone option to sync the time correctly? That fixed the time for me to the right time.

jasaw, 2019-06-20:
@burgerlX You get year 1970 on your device because it couldn't sync time. Please attach all log files here (do not copy and paste here). Which time sync protocol are you using?

nicemachines, 2020-05-21:
### TIME STILL NOT CORRECT** I have set motioneye to US/Eastern and actually gave the recommended SNTP the Colorado address 128.138.141.177
**  Do I have to actually reinstall motioneye because of some log that was deleted?  I say this because of a post above saying that is what happened to him.  WTF man?  I did have it working and now I can NOT set the time, keeps saying 1970 for the year!  Come on man, WTF?  Only thing I did before notice of this was I deleted pictures using the motioneye program (clicking on the second option to the right on top of the video on screen and getting there to view pictures, you can delete them there also, maybe there is a problem there making this program give us some really old date.  Again, WTF man!

starbasessd, 2020-05-21:
I use 'pool.ntp.org' and US Eastern. The only time I have an issue is when
there's no path to the Internet.
Which version of motionEyeOS are you using?
Are you able to watch the boot process, or copy the boot log here?
Where did you change your ntp.conf server?
THe default 'server 0.pool.ntp.org iburst' line works for me.
What log(s) did you delete? You normally run a script (logrotate) to back
them up prior to deleting the rotated scripts. THat can cause all kinds of
problems (which we've documented here before)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-21:
BTW, the above user posted part of his log, he didn't delete anything.

nicemachines, 2020-09-06:
Many months later and trying to get this time issue fixed and it just isn't doing it.  BTW, the post above with the log, he didn't delete anything, I did.  I mention this because the time was working correctly with the default setting, leaving the SNTP blank, like it's stated.  I deleted pictures and then the time went way back to the future.  1969,  It's frustrating and the correct time should be a priority to be fixed.  I am working with the PiZeroW now.  The Pi model shouldn't matter, shouldn't matter.   I am sad because nothing is being done about it.  I'll try the pool.ntp.org and making sure the setting above that entry is set to NTP.  At this point, with all the time I have spent trying to get this program to work correctly, years now on and off, it's just become to be a big waste of time for many!  If I know how to program, I would fix this, but I don't have the knowledge of the time.  Shame because this could be a great program.  Something like the Raspberry Pi's, you would think 64-bit RaspbianOS would be out, it's been what, 8 years?

nicemachines, 2020-09-06:
I correct myself, I did say he deleted stuff.  I stand corrected, but still I have to say the time, in my case, was working until I deleted pictures saved with motioneye.  Anyway.

starbasessd, 2020-09-06:
Can you please answer the questions I asked back in May?
Which version of motionEyeOS are you currently using?
Are you able to watch the boot process, or copy the boot log here?
Where did you change your ntp.conf server?
THe default 'server 0.pool.ntp.org iburst' line works for me.
What log(s) did you delete? You normally run a script (logrotate) to back
them up prior to deleting the rotated scripts. THat can cause all kinds of
problems (which we've documented here before)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

nicemachines, 2020-09-06:
I fixed it!  I tried everything and couldn't get it.  Then I erased the hostname I gave it and there ya go.  I guess if you give it a name for the host name it messes things up.  Then again, I was stupid and tried another browser to view (changed from EDGE to Opera)  When I went back to EDGE after viewing with Opera, the time is correct.  So it's either the internet browsers holding on to cache or it's the host name.   Leaving host name blank.  One or the other.  Got it narrowed down to this.  Sorry about being so blunt, sometimes I get way too upset and shouldn't.  I'll be back if this happens again, but I think I got it narrowed down.

starbasessd, 2020-09-06:
OK.
If/when you come back be sure to answer the questions.
Changing the hostname (unless you used invalid characters) shouldn't have
made any difference.
I use many different browsers in the process of supporting
motionEye/motionEyeOS, and other than Chrome, seldom have an issue.
Good luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-06:
Please close the ticket on github if you are working.

Eric-boop, 2021-03-09:
> OK. If/when you come back be sure to answer the questions. Changing the hostname (unless you used invalid characters) shouldn't have made any difference. I use many different browsers in the process of supporting motionEye/motionEyeOS, and other than Chrome, seldom have an issue. Good luck.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

This worked for me as well! Thanks!

closed 2021-03-09 by starbasessd

techny210, 2023-05-21:
I know is late but for those who stumble upon this with the same problem after few years, the fix is to ssh to the box (providing ssh is enabled_ and simply run: date -s "YYYY-MM-DD 20:45:00" or whatever time you want... Hope this helps someone... Cheers :-)