motioneye-project/motioneyeos #1958: IP address local address only

harisqazi1, 2019-06-19:
Hi, my problem is that my when i check my ip address on MotionEyeOS, it states http://10.0.0.41:8081. When I look at tutorials online for port forwarding, the IP addresses are all 192.xxx.xxx. I do not understand what I did wrong. I can access my camera when I type the http://10.0.0.41:8081 address, but in order to do port forwarding, I need to find the actual IP address. The image I have attached is seen when I click on "Streaming URL". What should I do to find the actual IP address?
[image]

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20190427.

### Board Model

I am using the following board/model: Raspberry Pi (c)2011.12 Model B Revision 2.0.

### Camera

I am using the following type of camera: Raspberry Pi Camera Module V2-8 Megapixel,1080p .

My camera model is: Raspberry Pi Camera Module V2-8 Megapixel,1080p.

### Network Connection

My motionEyeOS unit is connected to the network via: wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/3304090/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3304091/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3304092/motion.log)

jasaw, 2019-06-20:
Your local network has the IP address in the subnet of 10.x.x.x. You can follow the port forwarding guides and just replace the 192.168.x.x subnet with your 10.x.x.x subnet.

closed 2019-06-20 by jasaw

mrlowndes, 2020-04-22:
Hey so I have an issue like this. I was adding several cameras, all with IP addresses in the 10.0.0.10X range. I added a 4th and a 5th and they has an IP address of 192.168.101.XXX and I'm having loads of trouble connecting to those devices from my browser. Sometimes they connect, other times not. Any ideas?

starbasessd, 2020-04-22:
Need info:
what kind of connection? WiFi or Ethernet?
All close together or spread out?
Do you have multiple APs, or maybe multiple Routers? Sounds like you may
have more than one DHCP server on your network.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mrlowndes, 2020-04-22:
Wifi, spread out. We have a powerline ethernet extender which I think is causing the issues, but I read that they don't have DHCP servers on them

starbasessd, 2020-04-22:
Can you confirm by turning them off, or remoting into them?
Or move the ones that pull the odd addresses?
I'll bet you have one not set up correctly.

mrlowndes, 2020-04-22:
How would you suggest I set it up? Network IT is a bit of a dark art for me - totally bemused by it. The connection is being used right now so I can't change anything for a little bit

starbasessd, 2020-04-22:
Only have 1 router giving out DHCP addresses. All others should have it
turned off, especially on extenders. What devices are you using?

mrlowndes, 2020-04-22:
Hexicom https://images-na.ssl-images-amazon.com/images/I/A1tKrK5D-AL.pdf

mrlowndes, 2020-04-22:
I have the following options: DHCP, static, PPPOE, Bridge

starbasessd, 2020-04-22:
Bridge.

mrlowndes, 2020-04-23:
hey, this worked. Thanks so much

starbasessd, 2020-04-23:
No problem, please close your ticket.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mrlowndes, 2020-04-23:
erm.. how do I do that?

starbasessd, 2020-04-23:
Nevermind. I do most of my help through email. The ticket was not
re-opened, so you don't need to do anything, sorry.
Glad I was able to point you in the right direction.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?