motioneye-project/motioneyeos #1959: Where do i put python script?

Zeykox, 2019-06-19:
Hello,

I recently downloaded a python script thats allow to use discord webhooks but i don't know where to put it since i only have access to 2 folders while using Samba & FTP.

The creator of the script said to put it in /etc/data/ but i don't know how to access it.

So can someone tell me how? Thanks.

jasaw, 2019-06-21:
Copy the script to anywhere in your device. SSH into your device and move the script to /etc/data, something like this `mv /data/sdcard/myscript /data/etc`.

closed 2019-06-21 by jasaw

Zeykox, 2019-06-21:
[root@meye-b12d38a2 ~]# mv /data/sdcard/notify-discord.py /data/etc/
mv: can't rename '/data/sdcard/notify-discord.py': No such file or directory
[root@meye-b12d38a2 ~]#