motioneye-project/motioneyeos #1960: RPi-Zero panic rebooting (caused by network)

alexxtasi, 2019-06-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi Zero

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry Pi Camera Module 5MP Fisheye Lens

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * TP-LINK Ethernet-to-USB adapter

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. boot.log)
 ```
---- booting motionEyeOS 20190427 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20190427 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping ntpd: failed
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
---- booting motionEyeOS 20190427 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20190427 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping ntpd: failed
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
---- booting motionEyeOS 20190427 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20190427 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping ntpd: failed
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
---- booting motionEyeOS 20190427 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20190427 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping ntpd: failed
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
```

I tried MotionEyeOS with RPi3 model B and everything was OK
But when I used it with TP-LINK Ethernet-to-USB adapter or (this is the setup I am interested in) Pi Zero... cannot boot and shows error:
```
panic rebooting (caused by network)
```
Seems to me like MotionEyeOS cannot work with this ethernet adapter.
I am providing kernel messages when plug in the adapter

```
[  452.955775] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[  453.086786] usb 1-1.4: New USB device found, idVendor=2357, idProduct=0602
[  453.086795] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  453.086800] usb 1-1.4: Product: USB 10/100 LAN
[  453.086804] usb 1-1.4: Manufacturer: TP-LINK
[  453.086808] usb 1-1.4: SerialNumber: 503EAA5225A1
[  453.096878] cdc_ether 1-1.4:2.0 eth1: register 'cdc_ether' at usb-3f980000.usb-1.4, CDC Ethernet Device, 50:3e:aa:52:25:a1
[  453.097051] usbcore: registered new interface driver cdc_ether
```

any suggestions ?

thanks in advance

ccrisan, 2019-06-20:
Try downclocking your Pi Zero to 950 MHz or lower, in the Expert Settings section.

alexxtasi, 2019-06-21:
Hi @ccrisan and thanks for your suggestion.
Since my RPi Zero reboot always (cannot connect), I am trying to downclocking MotionEyeOS when running on RPi3 and then move the sd-card on to Zero to check if it works.

My problem is that in the _Expert Settings_ section I cannot find such a setting (_downclocking_)... only _GPU Memory_ .

Am I missing something here ?

ccrisan, 2019-06-22:
Don't use the RPi3 image on a Zero. They are not compatible.

alexxtasi, 2019-06-22:
Using RPi3 image on Zero and vice versa was my workaround (dummy I have to admit ! ), so I can copy log files in order to troubleshoot...

My problem now is : 
- how to use downclocking settings of RPi Zero (in _Expert Settings_ section) when it cannot boot and so I cannot use the web interface ?
- I did try editing ```/boot/config.txt``` file in sd-card, adding ```arm_freq=950```, so now this file is  :
```
arm_freq=950

gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128

dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
dtparam=audio=on
```
(correct me if this is not the right way to downclock Zero's CPU)

but again no success ... and reboot...
```
* Configuring wired network : no link
panic rebooting in 90 seconds (caused by network)
```

jasaw, 2019-06-24:
@alexxtasi I don't think your Ethernet issue is related to the Pi clock. Question is: Is the TP-LINK Ethernet-to-USB adapter supported on motionEyeOS? Can you attach your dmesg log here?

alexxtasi, 2019-06-24:
hey @jasaw 
Since when testing Zero with the adapter... it reboots, I used RPi3 with the same adapter (didn't work) and dmesg shows : 
```
[  452.955775] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[  453.086786] usb 1-1.4: New USB device found, idVendor=2357, idProduct=0602
[  453.086795] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  453.086800] usb 1-1.4: Product: USB 10/100 LAN
[  453.086804] usb 1-1.4: Manufacturer: TP-LINK
[  453.086808] usb 1-1.4: SerialNumber: 503EAA5225A1
[  453.096878] cdc_ether 1-1.4:2.0 eth1: register 'cdc_ether' at usb-3f980000.usb-1.4, CDC Ethernet Device, 50:3e:aa:52:25:a1
[  453.097051] usbcore: registered new interface driver cdc_ether
```

- any clue about supporting this eth-usb adapter ?
- is there a list with supported hardware (peripherals) by MotionEyeOS or some other source ?
- I am planning to test an ethernet HAT... should I check a hardware support list before buying ?
- in case of an unsupported hardware (eg HAT), is there a way to help support ? (providing info as dmesg or other) ?

thanks in advance

jasaw, 2019-06-25:
@ccrisan The USB-Ethernet adapter looks like is working, appeared as eth1. Which interface does the code look for when checking for network connectivity?

alexxtasi, 2019-06-25:
@jasaw , @ccrisan ... just to clarify:
the dmesg I provided above, is from RPi3 so the ```eth1``` might be misleading (cause of the existence of `eth0` maybe ?).
And I gave you this cause I don't know any other way of capturing logs from Zero when not booting at all (always rebooting with panic caused by network)

ccrisan, 2019-06-26:
First off, @alexxtasi my apologies: I forgot you can't overclock (or underclock) a Pi Zero and that's why you won't find those settings in the UI. The `config.txt` way that you chose was supposed to do the trick. If it didn't, than I'm out of ideas for now.

> * any clue about supporting this eth-usb adapter ?
> * is there a list with supported hardware (peripherals) by MotionEyeOS or some other source ?
> * I am planning to test an ethernet HAT... should I [...]

motionEyeOS uses Raspbian's kernel (source tree as well as configuration). So whatever Raspbian supports should work with motionEyeOS as well. When it comes to hats though, you might need to add some DTBs to the boot partition as well as to overlays in `config.txt`.

@jasaw 

> The USB-Ethernet adapter looks like is working, appeared as eth1. Which interface does the code look for when checking for network connectivity?

It's specified in `[/data]/etc/os.conf` as `OS_ETH` and defaults to `eth0`. It can be changed by setting the corresponding value in `/data/etc/os.conf`.

alexxtasi, 2019-07-04:
@ccrisan thanks for the suggestions,
I created a directory ```overlays``` inside boot partition, copied there the needed ```.dtbo``` files and then edited ```config.txt``` (```dtoverlay=enc28j60``` in my case).
Then, MotionEyeOS worked great with my ethernet hat !!

Should the wiki be more explanatory about this procedure ? Or it's my newbiesness ?? Anyway...

About the usb-ethernet adapter I cannot find any clue yet, but I can live with the hat ;) 

thanks for the great software !!

ccrisan, 2019-07-15:
Could you please try the latest [nightly dev build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and see if the WiFi problem persists?

alexxtasi, 2019-07-17:
In my setup I use the RPi Zero without WiFi (not the Zero W) and that's the reason why I used an ethernet hat (or the usb-to-ethernet adapter in the first place).

Since I found a solution for the hat (using Device Tree overlay as I described above), I should close this issue... or do you want me to test the nightly build for the `usb-eth adapter` issue ??

joesatriani10, 2020-02-26:
[this article] helps me on my raspberry pi 4 4GB (https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration)

joesatriani10, 2020-02-26:
this article help me with my raspberry 4 4gb [](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration)

closed 2020-03-02 by alexxtasi

jensenswitch, 2020-08-09:
I'm having the exact same problem, zero w but it looks for wired connection and panics when it cant find one..

starbasessd, 2020-08-09:
See my other response.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?