motioneye-project/motioneyeos #1963: ASUS Thinkerboard S unable to boot from internal storage

smogger, 2019-06-20:
### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-tinkerboard-20190427
But also tried: motioneyeos-tinkerboard-dev20190620

### Board Model

I am using the following board/model: ASUS Thinkerboard S REV 1.01

### Camera

Irrelevant for now

### Network Connection

Irrelevant for now

### Peripherals

Irrelevant for now

### Issue

Thinkerboard S unable to boot from internal storage.
No problem booting from SD card. 

Firmware installed from Etcher

ccrisan, 2019-06-26:
You could try replacing `mmcblk0` with `mmcblk1` in the following files:

 * `/boot/extlinux/extlinux.conf`
 * `/etc/fstab.disk`

But keep in mind though that a firmware update would overwrite those changes.

FroeMa, 2019-10-04:
Hello ccrisan,
very good project. But I have the same problem I tried to replace the mmc in that files. But they are write protected. I want to use my Tinker without SD card. So perhaps you can make a new release with using eMMC on Tinker. Or you have an other option to replace mmc.
Thank you

FroeMa, 2019-10-07:
Hello,
is there no solution?
BR
FroeMa

TechV, 2020-08-31:
So I don't have the S model, but I'm thinking about trying my hand at fixing the kernel issues in the tinkerboard builds for my own non-S board. As far as the S model goes, there is a jumper that is used to support standard images according to the documentation for that board. Have you given that method a try? Currently cameras based off the OV5647 (picam 1.3) chipset don't work, haven't tried an IMX cam yet.