motioneye-project/motioneyeos #1965: Add proper documentation for security

lukeswitz, 2019-06-22:


lukeswitz, 2019-06-22:
Ability to disable SMB etc. too much to even list. Probably create a secure branch when I get some time. The people deserve it

ccrisan, 2019-06-22:
@lukeswitz PRs are usually welcomed and there's always room for improvement, but your current attitude isn't very helpful, to be honest.

Users can disable services as they need, including SMB. If you think people would benefit more by a "security enhanced" motionEyeOS, feel free to fork it and provide such images yourself.

HTTPS isn't something that pertains to a device inside your network. It's something that you'd put at  the front somewhere, such as on a reverse-proxy frontend.

I agree with you that some docs on this matter would be useful, though. If you've got time to write an article/page about security issues and recommendations, I could add it to the wiki.

lukeswitz, 2019-06-22:
Loud and clear. Thanks for the thoughtful response

closed 2019-06-22 by lukeswitz

lukeswitz, 2019-06-27:
Append to the tweaks page a small section for firewall configuration perhaps? The additional ports I am unsure of at this time, but would appreciate any help. 

Here is what I came up with for use case local without SMB or NET-BIOS (open ports as needed)

```
#install and generate test output

sudo apt-get install ufw

sudo ufw disable
sudo ufw enable

#restrict to 8080/443 connection unless on local network

sudo ufw default deny incoming
sudo ufw default allow outgoing
sudo ufw allow from 192.168.0.0/24
sudo ufw allow 8080/tcp
sudo ufw allow https

#verify configuration, should appear as below 

sudo ufw status verbose
Status: active
Logging: on (low)
Default: deny (incoming), allow (outgoing), disabled (routed)
New profiles: skip

To                         Action      From
--                         ------      ----
443/tcp                    ALLOW IN    Anywhere
8080/tcp                   ALLOW IN    Anywhere
Anywhere                   ALLOW IN    192.168.0.0/24
443/tcp (v6)               ALLOW IN    Anywhere (v6)
8080/tcp (v6)              ALLOW IN    Anywhere (v6)

```